xyzroe/XZG #11: XZG Zigbee Firmware 20240510

rombrock, 2024-05-13:
Today I updated via LAN my ZigStar Lan Gateway.
Device model WT32-ETH01
FW revision 20240510
HW version ESP32-D0WD
The device is recognized by iobroker, but the connection is offline every 30 to 60 minutes. After a few minutes it is online again, but the adapter (zigbee.iobroker) in the iobroker remains yellow. Only after restarting the adapter will it turn green again.

rombrock, 2024-05-14:
Hello everyone, I reinstalled the old firmware and the connection is now stable again. No failure for over an hour, with the new version (FW revision 20240510) there were interruptions about every 45 minutes.
FW version: 0.6.10
Hardware: WT32-ETH01
ESP32 model: ESP32-D0WDQ5
I didn't change anything, no replacing the power supply.

Steltek, 2024-05-14:
Chiming in: I wasn't able to get FW 20240509 to work on my ZigStar LAN PoE gateway because the LAN connection would not work. It worked once immediately after flash with DHCP, and then never again. No combination of LAN DHCP or static IP address settings would make the LAN connection come up, and I tried re-flashing the firmware multiple times. (I also wasn't able to connect it to my Wi-Fi, though that may be normal as I've never used Wi-Fi before.) I rolled back to the old firmware and the device started working normally again.

Hardware and previous firmware:
FW version : 0.6.10
Hardware : TTGO T-Internet-POE
ESP32 model : ESP32-D0WDQ5

xyzroe, 2024-05-16:
> Today I updated via LAN my ZigStar Lan Gateway. Device model WT32-ETH01 FW revision 20240510 HW version ESP32-D0WD The device is recognized by iobroker, but the connection is offline every 30 to 60 [...]

You mean socket connection to Zigbee module, right?

xyzroe, 2024-05-16:
> Chiming in: I wasn't able to get FW 20240508/20240509 to work on my ZigStar LAN PoE gateway because the LAN connection would not work. It worked once immediately after flash with DHCP, and then never again. No combination of LAN DHCP or static IP address settings would make the LAN connection come up, and I tried re-flashing the firmware multiple times. (I also wasn't able to connect it to my Wi-Fi, though that may be normal as I've never used Wi-Fi before on it.) I rolled back to the old firmware and the device started working normally again.
> 
> Hardware and previous firmware: FW version : 0.6.10 Hardware : TTGO T-Internet-POE ESP32 model : ESP32-D0WDQ5

I need a little bit more information. 
1. Right after flashing it started ok, what device model was shown in web UI? 
2. If there isn't any connection for more that 60 seconds WiFi AP will start. Than you can check the device model and lan settings. 
3. Also check device model when LAN doesn't start successfully.

Steltek, 2024-05-16:
> I need a little bit more information.
>
>     1. Right after flashing it started ok, what device model was shown in web UI?
>
>     2. If there isn't any connection for more that 60 seconds WiFi AP will [...]

Right after upgrading, no LAN connected yet, only USB:

Device model: Olimex-ESP32-POE
FW revision: 20240510
HW version: ESP32-D0WD

Unplugged USB (power down) and plugged in Ethernet PoE. Device booted and took an IP from the DHCPd. Model info etc stayed the same.

Changing the network configuration to static (DHCP off, IP 192.168.1.x, gateway 192.168.1.y, DNS servers 1.1.1.1 and 8.8.8.8) and rebooting: No more LAN connection. (Orange LED on for 2-3 seconds, then green LED comes on and the orange LED is blinking for a second, back to orange LED on, .... loop) The XZG... AP appears but I either can't connect to it, or get disconnected immediately after connecting. (Feels like the device is in a boot/crash loop.)

xyzroe, 2024-05-18:
The only one way I think problems cloud be is board detecting mechanism. Just because of unplugged LAN cable during first boot. You could try to factory reset using tools page, with plugged in LAN cable. 
Also pay attention on device model on root page if any problems. Maybe all problems because of wrong board detection.

Steltek, 2024-05-20:
I flashed version 20240519 with via USB the network (without PoE) attached.
First reboot happened, and the device did not manage to connect to the network (Link on the switch did not come up, DHCP logged no request).
Factory reset did not change anything.

Disconnect USB and Ethernet, then plugged in Ethernet with PoE.
No link, no DHCP request.
Performed factory reset. Did not change anything.

The device remains stuck like before, the yellow Ethernet light is on for a few seconds, then the green one turns on and the yellow one blinks for a second, then the green one turns off and the yellow one stays on normally again (keeps looping like that).

Hardware in the WebUI still shows the same:

Device model: Olimex-ESP32-POE
FW revision: 20240519
HW version: ESP32-D0WD

Steltek, 2024-05-22:
My issue doesn't appear to match with what @rombrock reported above, so I've split it off into #16 .

rombrock, 2024-05-22:
> > Today I updated via LAN my ZigStar Lan Gateway. Device model WT32-ETH01 FW revision 20240510 HW version ESP32-D0WD The device is recognized by iobroker, but the connection is offline every 30 to 60 minutes. After a few minutes it is online again, but the adapter (zigbee.iobroker) in the iobroker remains yellow. Only after restarting the adapter will it turn green again.
> 
>     1. You mean socket connection to Zigbee module, right?
> 
>     2. How about uptime of the gateway? Does is drops to 0 after connection goes offline?

1. Yes, i mean socket Connection to Zigbee module
2. Since I have reinstalled the old firmware, I cannot say anything for sure about it

xyzroe, 2024-05-22:
It can be caused by two reasons: hang up Zigbee module or reboot ESP32. That's why we need to know about uptime.

rombrock, 2024-05-22:
Okay, then I'll update the second Zigbee adapter to the new version and see what happens. I'll report back.

rombrock, 2024-05-22:
I hope this helps a little. After 47 minutes, the Zigbee adapter lost its connection to the iobroker. The adapter (ESP32) then restarted and the connection times started again at zero. The connection to the iobroker is not established automatically. I cannot provide the raw data from the debug console as this is deleted when the adapter is restarted.

Another note: Since the Zigbee adapter did not work at the beginning after the web update, I installed the firmware via USB. Then the Zigbee adapter finally established the connection to the iobroker (green digit).
My File : configHw.json
{
"board": "WT32-ETH01",
"addr": 1,
"pwrPin": 16,
"mdcPin": 23,
"mdiPin": 18,
"phyType": 0,
"clkMode": 0,
"pwrAltPin": -1,
"btnPin": -1,
"btnPlr": 0,
"uartSelPin": -1,
"uartSelPlr": 0,
"ledModePin": -1,
"ledModePlr": 0,
"ledPwrPin": -1,
"ledPwrPlr": 0,
"zbTxPin": 17,
"zbRxPin": 5,
"zbRstPin": 33,
"zbBslPin": 32
}

rombrock, 2024-05-23:
I left the adapter running last night and could see that the ESP32 restarted after about 45 minutes. This morning I started the adapter again and after about 45 minutes the same thing happened again, the ESP32 restarted but the connection to the iobroker was not established automatically. A restart must be performed in iobroker. Here is the current information from the debug console.
58] -> fe 00 27 00 27
[29667] <- fe 12 67 00 00 43 2c d1 26 00 4b 12 00 00 00 07 09 02 90 a0 90 a0 b8
[29670] -> fe 04 25 05 00 00 00 00 24
[29671] -> fe 04 25 05 00 00 00 00 24
[29675] <- fe 12 67 00 00 43 2c d1 26 00 4b 12 00 00 00 07 09 02 90 a0 90 a0 b8
[29680] <- fe 01 65 05 00 61 fe 0b 45 85 00 00 00 00 00 05 05 04 03 02 01 cf fe 01 65 05 00 61
[29684] -> fe 09 24 00 06 09 01 05 00 00 00 00 00 26
[29685] -> fe 09 24 00 06 09 01 05 00 00 00 00 00 26
[29689] <- fe 0b 45 85 00 00 00 00 00 05 05 04 03 02 01 cf
[29694] <- fe 01 64 00 00 65 fe 01 64 00 b8 dd
[29696] -> fe 09 24 00 08 04 01 05 00 00 00 00 00 25
[29697] -> fe 09 24 00 08 04 01 05 00 00 00 00 00 25
[29704] <- fe 01 64 00 00 65 fe 01 64 00 b8 dd
[29708] -> fe 09 24 00 0a 04 01 05 00 00 00 00 00 27
[29709] -> fe 09 24 00 0a 04 01 05 00 00 00 00 00 27
[29715] <- fe 01 64 00 00 65 fe 01 64 00 b8 dd
[29719] -> fe 11 24 00 0b 04 01 00 04 00 00 02 01 05 0a 00 02 00 05 02 05 33
[29720] -> fe 11 24 00 0b 04 01 00 04 00 00 02 01 05 0a 00 02 00 05 02 05 33
[29729] <- fe 01 64 00 00 65 fe 01 64 00 b8 dd
[29732] -> fe 09 24 00 6e 04 01 05 00 00 00 00 00 43
[29733] -> fe 09 24 00 6e 04 01 05 00 00 00 00 00 43
[29739] <- fe 01 64 00 00 65
[29743] -> fe 09 24 00 0c 5e c0 05 00 00 00 00 00 ba
[29744] -> fe 09 24 00 0c 5e c0 05 00 00 00 00 00 ba
[29747] <- fe 01 64 00 b8 dd
[29751] <- fe 01 64 00 00 65 fe 01 64 00 b8 dd
[39767] | Connected client 5 from 192.168.1.75
[39768] -> ef
[40768] -> fe 00 21 01 20
[40775] <- fe 02 61 01 59 06 3d
[40778] -> fe 00 21 02 23
[40786] <- fe 0a 61 02 02 01 02 07 01 2a 8d 34 01 00 fc
[40790] -> fe 02 21 13 82 00 b2
[40802] <- fe 02 61 13 18 00 68
[40806] -> fe 04 21 1c 82 00 00 00 bb
[40828] <- fe 1a 61 1c 00 18 00 ee f4 94 8d 43 62 64 3e a9 1a 98 01 d6 2a e4 77 ee f4 94 89 13 00 00 56
[40831] -> fe 02 21 13 60 00 50
[40838] <- fe 02 61 13 01 00 71
[40841] -> fe 04 21 1c 60 00 00 00 59
[40850] <- fe 03 61 1c 00 01 55 2a
[40853] -> fe 02 21 13 21 00 11
[40862] <- fe 02 61 13 74 00 04
[40865] -> fe 04 21 1c 21 00 00 00 18
[40889] <- fe 76 61 1c 00 74 29 05 02 65 0f 65 00 1e 00 00 00 01 05 01 8f 00 07 00 02 0d 02 00 00 00 19 00 00 00 00 00 00 00 00 00 00 00 90 19 08 00 00 00 00 02 0f 0f 04 00 01 00 00 00 01 00 00 00 00 00 12 4b 00 21 9f b6 32 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 05 00 01 3c 0a 01 00 00 00 02 00 00 00 02
[40892] -> fe 02 21 13 62 00 52
[40900] <- fe 02 61 13 10 00 60
[40903] -> fe 04 21 1c 62 00 00 00 5b
[40913] <- fe 12 61 1c 00 10 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d 7c
[40918] -> fe 02 21 13 3a 00 0a
[40927] <- fe 02 61 13 11 00 61
[40930] -> fe 04 21 1c 3a 00 00 00 03
[40944] <- fe 13 61 1c 00 11 00 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d 7c
[40946] -> fe 02 21 13 3b 00 0b
[40955] <- fe 02 61 13 11 00 61
[40958] -> fe 04 21 1c 3b 00 00 00 02
[40972] <- fe 13 61 1c 00 11 00 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d 7c
[40976] -> fe 00 27 00 27
[40985] <- fe 12 67 00 00 43 2c d1 26 00 4b 12 00 00 00 07 09 02 90 a0 90 a0 b8
[40988] -> fe 04 25 05 00 00 00 00 24
[40996] <- fe 01 65 05 00 61
[41001] <- fe 11 45 85 00 00 00 00 00 0b 0c 6e 0b 0a 08 06 05 04 03 02 01 b6
[41004] -> fe 0b 24 00 0d 04 01 05 00 00 00 01 19 00 00 3a
[41013] <- fe 01 64 00 00 65
[41017] -> fe 09 24 00 2f 04 01 05 00 00 00 00 00 02
[41024] <- fe 01 64 00 00 65
[41027] -> fe 09 24 00 f2 e0 a1 05 00 00 00 00 00 9b
[41034] <- fe 01 64 00 00 65
[41037] -> fe 03 25 4a f2 84 0b 11
[41045] <- fe 13 65 4a 00 84 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b3
[... 62 lines cut ...]
[645930] <- fe 0e 45 84 00 00 00 00 00 08 0a 04 01 05 00 00 00 00 cd
[645933] -> fe 05 25 04 00 00 00 00 08 2c
[645937] <- fe 01 65 04 00 60
[645939] <- fe 0e 45 84 00 00 00 00 00 08 08 04 01 05 00 00 00 00 cf
[645942] -> fe 05 25 04 00 00 00 00 06 22
[645946] <- fe 01 65 04 00 60
[645948] <- fe 0e 45 84 00 00 00 00 00 08 06 09 01 05 00 00 00 00 cc
[645951] -> fe 05 25 04 00 00 00 00 05 21
[645957] <- fe 01 65 04 00 60 fe 0e 45 84 00 00 00 00 00 08 05 08 01 05 00 00 00 00 ce
[645961] -> fe 05 25 04 00 00 00 00 04 20
[645965] <- fe 01 65 04 00 60
[645967] <- fe 0e 45 84 00 00 00 00 00 08 04 07 01 05 00 00 00 00 c0
[645970] -> fe 05 25 04 00 00 00 00 03 27
[645974] <- fe 01 65 04 00 60
[645976] <- fe 0e 45 84 00 00 00 00 00 08 03 04 01 05 00 00 00 00 c4
[645981] -> fe 05 25 04 00 00 00 00 02 26
[645985] <- fe 01 65 04 00 60
[645987] <- fe 0e 45 84 00 00 00 00 00 08 02 01 01 05 00 00 00 00 c0
[645992] -> fe 05 25 04 00 00 00 00 01 25
[645996] <- fe 01 65 04 00 60
[645998] <- fe 0e 45 84 00 00 00 00 00 08 01 04 01 05 00 00 00 00 c6
[646004] -> fe 00 25 50 75
[646009] <- fe 18 65 50 00 00 09 90 19 00 00 00 12 4b 00 21 9f b6 32 00 00 00 00 00 00 00 00 19 d7
[646028] -> fe 02 21 0f 00 00 2c
[646031] <- fe 01 61 0f 00 6f

xyzroe, 2024-05-23:
But you uptime doesn't drops to zero, it is only goes up. 
How did you detect that ESP32 restarted?
It looks like Zigbee chip hangups. What Zigbee chip are you using? What firmware version?

rombrock, 2024-05-23:
But your operating time doesn't go down to zero, it just goes up.
How did you detect that ESP32 was restarted?
The Device operating time counted from zero again.

It looks like the Zigbee chip is stuck. What Zigbee chip are you using? 
That's an ebyte chip.

What firmware version? 
FW revision 20221226

xyzroe, 2024-05-24:
> The Device operating time counted from zero again.
> 
But in logs that you provide time only goes up. So I thought that way.

> That's an ebyte chip.
> 
OK. 

> FW revision 20221226
> 
> I have no problems with firmware version 06.10.
OK. But we need to check all first. 

Have you tried to use latest version of XZG ?

xyzroe, 2024-05-24:
Do you keep your tab with web UI open continuously?

rombrock, 2024-05-24:
Yes, the tab is permanently open.

When the failure occurs, 5 devices are shown as connected. But since it is a test environment, it can actually only be one device. Then in iobroker we noticed that the RAM usage was no longer 130 MB but had gone up to over 600 MB.

xyzroe, 2024-05-24:
It looks like some problems with iobroker. 
It's the first report about the socket connections count is wrong.

xyzroe, 2024-05-24:
I have no other ideas at the moment.

rombrock, 2024-05-27:
That's not a problem at the moment. My active Zigbee LAN adapter is currently still running on the old version.

commit 2550e564 (2024-06-06): 20240607

### Zigbee Updates
- Add Zigbee role on root page and MQTT
- Display saved Zigbee version if unable to detect the installed version (Web UI and MQTT)
- Add check for available Zigbee updates
- Fix Zigbee version detection after Zigbee firmware update
- Add **automatic change of baud rate** after installing new Zigbee firmware
- Add NV parameters for last installed Zigbee firmware version and role
- Add MQTT sensor for Zigbee update availability
- HTTP API command to update Zigbee firmware now uses the role and gets the latest firmware from GitHub
- Remove Zigbee firmware check during [... cut]
closed 2024-06-06 by xyzroe