xyzroe/XZG #15: Warning in HA about the missing attribute wlanSsid in a template

mihsu81, 2024-05-21:
**Describe the bug**
Getting warnings in HA when using Ethernet without WiFi:
Template variable warning: 'dict object' has no attribute 'wlanSsid' when rendering '{{ value_json.wlanSsid }}'

**To Reproduce**
Steps to reproduce the behavior:
1. Only connect over Ethernet
2. Check HA logs

**Expected behavior**
Shouldn't get these warnings while only connected over Ethernet.

**Screenshots**
```
Logger: homeassistant.helpers.template
Source: helpers/template.py:2613
First occurred: 20 mai 2024 la 20:18:52 (1427 occurrences)
Last logged: 19:56:18

Template variable warning: 'dict object' has no attribute 'wlanSsid' when rendering '{{ value_json.wlanSsid }}'
```

**Additional context**
Olimex-ESP32-POE
20240519

xyzroe, 2024-05-22:
Thank. I'll fix it soon.

closed 2024-05-25 by xyzroe

polenco, 2025-05-07:
The behaviour as desribed above is still there or is re-occuring. Using Home Assistant Green with ethernet only. 

Core 2025.4.4
Supervisor 2025.04.1
Operating System 15.2
Frontend 20250411.0

xyzroe, 2025-05-07:
> The behaviour as desribed above is still there or is re-occuring. Using Home Assistant Green with ethernet only. 
> 
> Core 2025.4.4
> Supervisor 2025.04.1
> Operating System 15.2
> Frontend 20250411.0

what's about XZG? What version do you use?

polenco, 2025-05-08:
Thanks for your swift reply. I should have known based upon the name of the github repository, but untill your reply I was unware the ZigBee hub could have been involved in these warnings. 

Below is the info of my ZigBee Hub which is indeed running the XZG firmware. Under the "Status" tab this info is shown about the device:
- Device model	UZG-01
- FW revision	20240519
- HW version	ESP32-D0WD-V3
- Processor	2 cores @ 160 MHz
- Heap	       103 / 316 KiB
- Memory	4 Mb, external
- NV storage	200 / 630 entries
- File system	16 / 1408 KiB

ZigBee:
HW version	CC2652P7

In the debug console of Debug Tools I get this info when checking the ZigBee version:
[783188428] | [ZB] fw: 20230507

I now notice that the date of the FW revision is earlier than the date of 25th of may 2024 when you closed this issue as complete. So most likely I'II need to update this firmware? And in that case should I shutdown HA and then take the ZigBee Hub to a location where no active ZigBee devices (belonging to the network) are in the vicinity before performing a update? This because I see the warning on the ZigBee tab: "Changing the firmware on a running network cause the network break down!" Although this might only be applicable for changing the role of the ZigBee Hub.

xyzroe, 2025-05-09:
You need to update the ESP32 firmware, not the Zigbee (CC2652P*) firmware. That won't cause any problems with your network. It will only be down for a few minutes.

polenco, 2025-05-12:
Thank you very much again for your help on a already close issue.
For those facing the same problem, below  is what I did for updating the ESP32 firmware to version 20240914 using USB connection as the install from the WEB GUI was not installing.
In my case I have a UZG-01 Zigstar gatewat.

- Connect the USB-C port to a computer. In my case a windows laptop with Windows 11.
- Pre-install the driver for the CH304C USB chip which is inside the UZG-01 Zigstar gateway
  Source description UZG-01 Zigstar: https://uzg.zig-star.com/product/
  Link to Windows driver for the CH340C USB chip: https://www.wch.cn/download/file?id=5
  Unpack the zipfile and run the setup.exe. If you do not trust a website with chinese characters then goto:
  https://sparks.gogo.co.nz/ch340.html
- Launch a Chrome Browser, 
- Go to: https://xzg.xyzroe.cc/install/
- Select your firmware version
- Install
- Wait a few minutes and your done.