xyzroe/XZG #16: LAN connection does not work with ZigStar LAN PoE gateway.

Steltek, 2024-05-22:
Splitting this off into a separate issue as it doesn't appear to match what @rombrock reported in theirs:

I wasn't able to get FW 20240508/20240509 to work on my ZigStar LAN PoE gateway because the LAN connection would not work. It worked once immediately after flash with DHCP, and then never again. No combination of LAN DHCP or static IP address settings would make the LAN connection come up, and I tried re-flashing the firmware multiple times. (I also wasn't able to connect it to my Wi-Fi, though that may be normal as I've never used Wi-Fi before on it.) I rolled back to the old firmware and the device started working normally again.

Hardware and previous firmware:
FW version : 0.6.10
Hardware : TTGO T-Internet-POE
ESP32 model : ESP32-D0WDQ5

_Originally posted by @Steltek in https://github.com/xyzroe/XZG/issues/11#issuecomment-2109620471_

xyzroe, 2024-05-22:
You could try to edit configHw.json manual. 

```
{
  "board": "T-Internet-POE",
  "addr": 0,
  "pwrPin": -1,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": -1,
  "btnPin": -1,
  "btnPlr": 0,
  "uartSelPin": -1,
  "uartSelPlr": 0,
  "ledModePin": -1,
  "ledModePlr": 0,
  "ledPwrPin": -1,
  "ledPwrPlr": 0,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}
```

[image]

If it would help I will know what I need to fix.

xyzroe, 2024-05-23:
You are using [LilyZig](https://zig-star.com/projects/zigstar-lilyzig/), right?

Steltek, 2024-05-23:
> You are using [LilyZig](https://zig-star.com/projects/zigstar-lilyzig/), right?

No, mine looks more like this: https://uzg.zig-star.com/
Here's the original Amazon reference: https://www.amazon.de/-/en/gp/product/B0CS39GHH7/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

Steltek, 2024-05-23:
Manually modifying the configHw.json made it work IF I connect it to PoE. When connected to USB with a non-PoE ethernet cable, the old issue persists. (Not a problem for me though.)

xyzroe, 2024-05-23:
You need to move dip switch on the board to use USB power mode.

Steltek, 2024-05-23:
> You need to move dip switch on the board to use USB power mode.

I don't have access to the board (unless I crack the case open), but it's no problem, I run the device with PoE anyway.

closed 2024-05-23 by Steltek

xyzroe, 2024-05-24:
> > You are using [LilyZig](https://zig-star.com/projects/zigstar-lilyzig/), right?
> 
> No, mine looks more like this: https://uzg.zig-star.com/ Here's the original Amazon reference: https://www.amazon.de/-/en/gp/product/B0CS39GHH7/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

So I not sure about configHw.json I provided earlier. Usually Hamgeek devices use the same pins as UZG-01. But in you case it have some differences. But if you would the working way - it's the most important.

Steltek, 2024-05-24:
With the newfound knowledge of USB mode not working by default, which may have compromised my previous testing, I have tested a bit more:

1. I did the OTA firmware upgrade and bam, the problem was back. The device needed a power cycle to start working again.
2. I rolled back the configHw.json:

```
{
  "board": "Olimex-ESP32-POE",
  "addr": 0,
  "pwrPin": 12,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": -1,
  "btnPin": -1,
  "btnPlr": 0,
  "uartSelPin": -1,
  "uartSelPlr": 0,
  "ledModePin": -1,
  "ledModePlr": 0,
  "ledPwrPin": -1,
  "ledPwrPlr": 0,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}
```
I saved and restarted the device via the web UI... and the problem was back again. I power cycled the device (PoE off and back on) and it just started working again.

So I think the problem isn't the PIN out definition per se (both seem to work) but the fact that the device simply needs a hard power cycle to function after I save the settings. (Plus me doing testing in USB power mode without realising that that requires a DIP switch inside to be flipped.)

charrus, 2024-05-25:
I may have the same issue on a the same device. I noticed the .local address was 192.168.1.1, but its real IP address came from the DHCP server in the correct subnet (172.16.17.x).

Connecting to the DHCP address resolved the issue for me.

My configHW.json:
```
{
  "board": "UZG-01",
  "addr": 0,
  "pwrPin": -1,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": 5,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 33,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}
```

Steltek, 2024-05-25:
After the update from 20240525.1 to 20240525.1.1, the connection issues re-appeared so I've swapped out the Olimex config for the T-Internet-POE one again, which seems to work.

```
{
  "board": "T-Internet-POE",
  "addr": 0,
  "pwrPin": -1,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": -1,
  "btnPin": -1,
  "btnPlr": 0,
  "uartSelPin": -1,
  "uartSelPlr": 0,
  "ledModePin": -1,
  "ledModePlr": 0,
  "ledPwrPin": -1,
  "ledPwrPlr": 0,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}
```