xyzroe/XZG #30: Redirection

chrizy66, 2024-05-25:
After update from 0.2.6 can't access web ui by fqdn; opening http://uzg-01/ redirects to 192.168.1.1 but my local address pool is in 10.0.0.0 range; Chrome, Firefox, Edge

xyzroe, 2024-05-25:
Have you tried xzg.local ?

chrizy66, 2024-05-25:
Tried:

http://uzg-01/, http://uzg-01.local/, http://uzg-01.home.arpa/ (my internal domain)

all leads to redirection to 192.168.1.1

If i try

http://xzg.local/, http://xzg.home.arpa/, http://xzg/

page not found.

xyzroe, 2024-05-25:
Please try new release

chrizy66, 2024-05-25:
Nothing changed on 20240525.1; still redirection to 192.168.1.1 after a while.
nslookup shows correct data:
Name:    uzg-01.home.arpa
Address:  10.xx.xx.19

xyzroe, 2024-05-25:
@chrizy66  try now

charrus, 2024-05-25:
This resolved DHCP address/DNS/mDNS/static

chrizy66, 2024-05-25:
All is fine on the version 20240525.1.1

closed 2024-05-25 by xyzroe