xyzroe/XZG #42: Прошивка ESP32 не обновляется через веб интерфейс

Sergey8100, 2024-05-27:
Доброго времени суток. Постоянно получаю уведомление о доступности новой прошивки, но при попытки обновления постоянно висит в статусе "Скачивание последнего релиза GitHub...". Пробовал перезагружать роутер и координатор, ничего не помогает.

Sergey8100, 2024-05-27:
Good day. I constantly receive a notification about the availability of new firmware, but when I try to update, it always hangs in the status “Downloading the latest GitHub release...”. I tried rebooting the router and coordinator, nothing helps.

xyzroe, 2024-05-27:
Thanks for bug report. I already know about that bug. Working to solve it. 
You can try to reboot gateway and try again just after reboot. 
Or use local file update option.

koeberlue, 2024-05-28:
As a workaround setting the timezone to `Europe/Kiev` and then restarting when prompted solved the issue for me. 

(Maybe it was just the restart that helped)

Sergey8100, 2024-05-28:
Rebooting didn't help me. I updated via file.

koeberlue, 2024-05-29:
When I looked at the Status page for some time I noticed that the entry for the DNS server is changing constantly. 
Strangely this behavior somehow seems to be connected to the "Local time".

Every time the local time shows a full minute (e.g. Wed May 29 14:51:**00** 2024) the DNS server entry will show `192. 168.188.3` (which is the one my router promotes via DHCP).

Then exactly after 19 seconds (e.g. Wed May 29 14:51:**19** 2024) the DNS server entry will show `253.0.0.0` for the rest of the minute.

So for the firmware update to succeed I have to wait until a full minute starts, then I have 19 seconds to start the update.

Maybe someone more knowledgeable about the firmware can draw the correct conclusions from this.

xyzroe, 2024-05-29:
Your router renews DHCP lease every minute. So your device drops DNS to 253.0.0.0 (bug of current version of LWIP) every minute. XZG runs mechanism to restore original DNS servers received from DHCP also every minute. 
You need to increase your DHCP lease time for XZG in your router.

xyzroe, 2024-05-29:
Hm.. Interesting. 
I don't know any way to capture network packages on the ESP. 
I have to surf the net for more information about this problem.

koeberlue, 2024-05-29:
I did some more debugging and I share your suspicion that something on my network is causing this behavior.

What caught my eye is the fact, that the DNS server is always reset to `253.0.0.0` at the *beginning of the 19th* second of every minute. I briefly checked the code in this repository but I could not find anything that should be doing anything special at that second. 

I will dump some things I discovered, maybe this might help painting the bigger picture:

* A cron job within the firmware resets the DNS entry every full minute
  * Sometimes this causes a short disconnect on the Zigbee client
* In my case the DNS server is reset to `253.0.0.0` at every 19th second of every minute
  * In case some one else has the same problem, I would love to know, if it is the same second or another
* I see the same behavior with ethernet, WiFi or both
  * The DNS server is reset for both, ethernet and WiFi, simultaniously
* Disabling DHCP and hard coding the DNS server does **not** help. The DNS server is still set to `253.0.0.0`

Looking through the wireshark dump from the WiFi-Extender the device was connected to, I found these packages, which were sent at the 19th seconds on both captures I made:

|No. | Time | Source | Destination | Protocol | Length | Info | 
| --- | --- | --- | --- | --- | --- | --- |
|523 | 16:55:19,242220 | fe80::c52c:4792:8724:1e13	ff02::1 |ICMPv6  | 104  | Neighbor Advertisement <redacted>:2df4:f9c4:ba15:279f (ovr) is at dc:a6:32:de:e5:fb |
| 524 |16:55:19,242848 |f e80::dea6:32ff:fe60:a3ec | ff02::1 |ICMPv6 |104 |Neighbor Advertisement <redacted>:dea6:32ff:fe60:a3ec (ovr) is at dc:a6:32:60:a3:ec | 
| 525 |16:55:19,251772 | :: |ff02::1:ff00:0 | ICMPv6 |96 | Neighbor Solicitation for <redacted>:200:ff:fe00:0 |
| 526 |16:55:19,252019 |:: |ff02::1:ff00:0 | ICMPv6 | 96 | Neighbor Solicitation for <redacted>:ff:fe00:0 |
| 527 | 16:55:19,253679 | fe80::5054:ff:fe6d:72bd | ff02::16 | ICMPv6 | 108 | Multicast Listener Report Message v2 |
| 528 |16:55:19,256590 | fe80::aaa1:59ff:fee7:43ef | ff02::16 | ICMPv6 |108 |Multicast Listener Report Message v2 |

So there are some IPCMPv6 broadcasts that might cause trouble. 

I also realized this project is using an external network library so it dug a little bit into it and found this issue, that sound somewhat similar: https://github.com/espressif/arduino-esp32/issues/7602 It was resolved just a few weeks ago, so maybe an update of this library might help?

@xyzroe Would it be feasible to replace some of the `LOGD` statements with `printLogMsg`? Right now the debug console is mostly flooded with binary data (from the Zigbee chip?). So I guess some network messages might not interfere with the log while they would provide us with some useful info about what is going on.

koeberlue, 2024-05-29:
> @xyzroe Would it be feasible to replace some of the `LOGD` statements with `printLogMsg`? Right now the debug console is mostly flooded with binary data (from the Zigbee chip?). So I guess some [...]

Ah, you just did this an hour ago. Will check the logs. Thank you!

koeberlue, 2024-05-29:
Here are the logs. I restarted the ESP to capture the start up phase. 

```
[1218] | [ZB] Chip: CC2652P7
[1218] | [ZB] IEEE: _redacted_
[1229] | [ZB] Flash size: 704 KB
[3587] | [Time] Wed May 29 20:01:57 2024
[3594] | [DNS] Saved ETH - 192.168.188.3
[3640] | [Time] Wed May 29 20:01:57 2024
[3647] | [DNS] Saved ETH - 192.168.188.3
[3958] | [ZB] FW: 20240316
[11714] | Connected client 1 from 192.168.188.61
[...]

// Then this message repeats (I think i missed the first invocations)

[126046] | [Time] Wed May 29 20:04:00 2024
[126046] | [DNS] Restored ETH - 192.168.188.3
[126053] | [DNS] Saved ETH - 192.168.188.3
[...]
[186060] | [Time] Wed May 29 20:05:00 2024
[186061] | [DNS] Restored ETH - 192.168.188.3
[186067] | [DNS] Saved ETH - 192.168.188.3
```

xyzroe, 2024-05-30:
#56

xyzroe, 2024-06-03:
Could you check on the latest release?

koeberlue, 2024-06-04:
Same behavior on the latest release

xyzroe, 2024-06-04:
Web UI notification checks update using your browser. But ESP fails to download the update. You should check DNS on the root page before trying to update.

koeberlue, 2024-06-05:
> Web UI notification checks update using your browser. But ESP fails to download the update. You should check DNS on the root page before trying to update.

Yeah, that is absolutely what's happening. And this is really not such a big issue, because nothing bad happens if the download fails.

But it does become a bigger issue when  installing the Zigbee OTA update, because it erases the chip before downloading. So when the download fails, the chip will be erased without a new firmware being installed. Also this process is harder to time correctly, because the download will not start immediately after pressing the install button.

So I am still wondering why the DNS config is reset every minute. I will try debugging the issue locally, but this might take some weeks to block some time off.

closed 2024-06-06 by xyzroe

koeberlue, 2024-06-07:
I can confirm both ESP32 and Zigbee update now work for me in `20240607.1` without any problems.

Also the `Status` page now always shows the correct DNS server, which seems a little bit strange, since nothing in the changelog should have fixed this directly (?).

On a side note: Connecting to MQTT does no longer work using the hostname of my mosquitto instance. But it works perfectly using the IP address  of the server directly. I will check if it a local issue in my home network and will create a new issue if I can pin down the reason for this.

xyzroe, 2024-06-08:
I think that I know reason. The same problem with DNS. Look change log for 20240607. I fixed the DNS in memory, and use it everywhere. But before all DNS related action I run check DNS in memory with current one, and fixed if need.

commit 70fddbd6 (2024-05-28): 20240528.1

- Add 230400 & 460800 baud for ZB socket
- Fix DNS replace for trash while DHCP update, it could fix #42, #43