xyzroe/XZG #43: UZG-01 - Git CC2652 update not updating via web interface

GD-Dal, 2024-05-28:
I realized I was running an older version of the zigbee firmware (20230507).
So I wanted to update it via the web gui.
[image]
[image]
But when I click install, it says:
[image]
and it hangs there for a while.
Then it simply states "failed"
Is there another way to flash it?

Thanks

koeberlue, 2024-05-28:
I am running into the same issue. Here are the logs from the debug console:

```
[174609] | Start Zigbee flashing
[176924] | [ZB] fw: unknown!
[176925] | Installing from https://raw.githubusercontent.com/xyzroe/XZG/zb_fws/ti/coordinator/CC1352P7_coordinator_20240316.bin
[180088] | Erase completed!
[182046] | Error initializing flash process
[183047] | Erase completed!
[183462] | Failed to download file, HTTP code: -7

[184971] | Erase completed!
[186886] | Error initializing flash process
[187888] | Erase completed!
[188013] | Failed to download file, HTTP code: -7

[189514] | Erase completed!
[191426] | Error initializing flash process
[192428] | Erase completed!
[192615] | Failed to download file, HTTP code: -7

[194116] | Erase completed!
[196033] | Error initializing flash process
[197035] | Erase completed!
[197082] | Failed to download file, HTTP code: -7

[198583] | Erase completed!
[200550] | Error initializing flash process
[201552] | Erase completed!
[201599] | Failed to download file, HTTP code: -7

[203100] | Erase completed!
[205068] | Error initializing flash process
[206070] | Erase completed!
[206117] | Failed to download file, HTTP code: -7

[207618] | Erase completed!
[209527] | Error initializing flash process
[210529] | Erase completed!
[210575] | Failed to download file, HTTP code: -7

[212076] | Erase completed!
[214041] | Error initializing flash process
[215043] | Erase completed!
[215231] | Failed to download file, HTTP code: -7

[216732] | Erase completed!
[218613] | Error initializing flash process
[219615] | Erase completed!
[219662] | Failed to download file, HTTP code: -7

[221163] | Erase completed!
[223131] | Error initializing flash process
[224133] | Erase completed!
[224181] | Failed to download file, HTTP code: -7

[225231] | Failed to flash Zigbee
```

Possibly related to #42?

xyzroe, 2024-05-28:
You could try to restart ESP32 and try again. 

The other option is to use ZigStarGW-MT 
Multitool

GD-Dal, 2024-05-28:
> ZigStarGW-MT
> Multitool

I resolved using this

closed 2024-06-03 by xyzroe