xyzroe/XZG #44: PoE powered UZG-01 does not come online after reboot

GD-Dal, 2024-05-28:
Not sure if this is related to the switch or the UZG-01
If I restart the UZG-01, it does not come back until I shutdown the switchport and turn it back on.
Until then, the UZG network port blinks for a second, then goes dark. Blinks for a second then goes dark, etc.

The switch is a Cisco Catalyst 9300 of model C9300L-48UXG-4X
The port settings looks like this:
```
interface GigabitEthernet1/0/31
 description UZG
 switchport access vlan 1098
 switchport mode access
 device-tracking
 speed 100
```

The configHW.json debug file contains this:

```
{
  "board": "UZG-01",
  "addr": 0,
  "pwrPin": -1,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": 5,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 33,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}
```

GD-Dal, 2024-05-29:
[image]

I don't know how to open the box, though. It looks solid

xyzroe, 2024-05-29:
Ok. No need. I'll check and come back.

GD-Dal, 2024-05-30:
Didn't help at all.
But I discovered one thing though.

I turned of power on the switchport and powered up the UZG-01 via USB-cable instead.
And when I rebooted the UZG-01 some time later, the same thing happened.

The network port just blinked as before until I unplugged the usb cable, waited 10 seconds, and replugged it.

This is becoming the most unstable piece of hardware I have purchased for a long time.

Not possible to reboot without hassle
LQI is low on all zigbee devices.
Zigbee2MQTT crashes regularily because loss of connection

NeoAcheron, 2024-05-31:
I am having similar problems with my UZG-01. Funny thing is that I updated the firmware two nights ago from 0.2.6, which was running for months without issues, but the new XZG firmware stays up for a few hours then dies, and basically my entire Zigbee network is down.

I'll be moving back to the old firmware while these issues are resolved.

xyzroe, 2024-06-03:
> Didn't help at all. But I discovered one thing though.
>
> I turned of power on the switchport and powered up the UZG-01 via USB-cable instead. And when I rebooted the UZG-01 some time later, the same [...]

Let's separate Zigbee and ESP issues. 
Problem with LAN port is ESP issue. 
But low LQI as well as loss of Zigbee connection are Zigbee releated issues. Find more info in Zigbee firmware repository.

I will look for some solution to made ESP32 works with LAN port more stable.

xyzroe, 2024-06-03:
> You could try to change `"pwrAltPin": 5,` to `"pwrAltPin": -1,` in your configHW.json

You could try to change "pwrAltPin": 5, to "pwrAltPin": -1, in your configHW.json. Have you?

xyzroe, 2024-06-03:
> I have the same issue with UZG-01 (POE-0.4)

You could try to change "pwrAltPin": 5, to "pwrAltPin": -1, in your configHW.json. Have you?

xyzroe, 2024-06-03:
@mercenaruss could you help us ?

mercenaruss, 2024-06-04:
@tonius7570 @zoptooniek Try to change "pwrAltPin": -1, to "pwrAltPin": -1, in your configHW.json. 
If any of you have some time,please drop me an email support@zig-star.com and we will try remotely to debug this.

xyzroe, 2024-06-08:
Could you try factory reset after updating to 20240607*

xyzroe, 2024-06-08:
@zoptooniek could you check it later to confirm, and if no issues - close this issue ?

commit 2550e564 (2024-06-06): 20240607

### Zigbee Updates
- Add Zigbee role on root page and MQTT
- Display saved Zigbee version if unable to detect the installed version (Web UI and MQTT)
- Add check for available Zigbee updates
- Fix Zigbee version detection after Zigbee firmware update
- Add **automatic change of baud rate** after installing new Zigbee firmware
- Add NV parameters for last installed Zigbee firmware version and role
- Add MQTT sensor for Zigbee update availability
- HTTP API command to update Zigbee firmware now uses the role and gets the latest firmware from GitHub
- Remove Zigbee firmware check during [... cut]
closed 2024-06-09 by xyzroe