xyzroe/XZG #50: Support HMG-01 POE Plus

radioelf, 2024-05-29:
Hello,
**Is your feature request related to a problem? Please describe.**
Support for HMG-01 POE Plus (Chinese clone)

**Describe the solution you'd like**
Is it possible to support the HMG-01 POE Plus device (Chinese clone)? I believe it is a "simple" modification?. In the firmware https://github.com/mercenaruss/uzg-firmware, I changed #define ETH_POWER_PIN_1 -1 to #define ETH_POWER_PIN_1 5 and it seems to be working correctly for the past couple of months. I see that this new firmware supports several devices: UZG-01, SLZB-06, Olizig PoE, LilyZig PoE, ZigStar LAN. Could HMG-01 POE Plus be included? 

**Describe alternatives you've considered**
Change of the equivalent code in the XZG firmware regarding the use of ETH_POWER_PIN_1, following the modification from #define ETH_POWER_PIN_1 -1 to #define ETH_POWER_PIN_1 5 as done in the UZG firmware, and I do not know if further changes would be necessary.

**Additional context**
Thank you.

xyzroe, 2024-05-29:
In XZG and UZG config is the same. 
```
#define ETH_POWER_PIN_ALTERNATIVE_1 5
#define ETH_POWER_PIN_1 -1
```
https://github.com/mercenaruss/uzg-firmware/blob/813589ad587cbe860589a4334f002e7351525ed1/src/config.h#L11
https://github.com/mercenaruss/uzg-firmware/blob/813589ad587cbe860589a4334f002e7351525ed1/src/config.h#L12

and in XZG
`{.addr = 0, .pwrPin = -1, .mdcPin = 23, .mdiPin = 18, .phyType = ETH_PHY_LAN8720, .clkMode = ETH_CLOCK_GPIO17_OUT, .pwrAltPin = 5},  // 2 T-Internet-POE`
https://github.com/xyzroe/XZG/blob/31ff9e3332f6fcbdd392d358bd7985208859f902/src/const/hw.cpp#L8
`{"UZG-01", .ethConfigIndex = 2, .zbConfigIndex = 0, .mistConfigIndex = 1},`
https://github.com/xyzroe/XZG/blob/31ff9e3332f6fcbdd392d358bd7985208859f902/src/const/hw.cpp#L35

xyzroe, 2024-05-29:
Could you check and give a feedback?

radioelf, 2024-05-30:
Hi, 
I have tried to compile the latest version of the repository, but I am getting some errors during the compilation:

src/etc.cpp:20:10: fatal error: WireGuard-ESP32.h: No such file or directory

I added ciniml/WireGuard-ESP32@^0.1.5 in platformio.ini, but I get the following error:
src/etc.cpp: In function 'void wgBegin()':
src/etc.cpp:722:29: error: no matching function for call to 'WireGuard::begin(IPAddress&, IPAddress&, uint16_t&, IPAddress&, char [50], char [50], char [50], uint16_t&, IPAddress&, IPAddress&, bool&, const char*&)'
             wg_preshared_key))

xyzroe, 2024-05-30:
You don't need to compile a custom build. Just use the bin from releases.

xyzroe, 2024-05-30:
You can add any custom device by editing configHw.json in Tools/Debug/File browser after installing XZG firmware.

xyzroe, 2024-05-30:
Can you please test and provide 100% working config for HMG-01 POE Plus?

radioelf, 2024-05-31:
This is the information I have been able to obtain from the hardware:

--- clone UZG-01 HamGeek Corogoo HMG-01 POE Plus---
SMC8720A Lan
RS2099  bidirectional 4-channel single-pole (SPDT)
CH340C Usb-Rs232
GPIO CONTROL SMC8720A-LAN8720
ETH_CLK_MODE_1 ETH_CLOCK_GPIO17_OUT GPIO 3
ETH_POWER_PIN_1 5
ETH_TYPE_1 ETH_PHY_LAN8720 Type of the Ethernet PHY
ETH_ADDR_1 0 
ETH_MDC_PIN_1 GPIO 23
ETH_MDIO_PIN_1 GPIO 18
CC2652P
GPIO CONTROL CC2652P
CC2652P_RST OUTPUT GPIO 16
CC2652P_FLSH OUTPUT GPIO 32
CC2652P_RXD GPIO 36
CC2652P_TXD GPIO 4
--------------------------
GPIOs ESP32
GPIO EN-> RESET ->CH340
GPIO 0->  BOOT ->CH340
GPIO 1->  RXD Pin 13 (NC4) RS2099 (SPDT) USB
GPIO 2->  NC
GPIO 3->  RXD pin 9 (NC3) RS2099 (SPDT) USB
GPIO 4->  TXD pin 16 COM1 RS2099 
GPIO 5->  PIN 15 nRST LAN8720  
GPIO 6->  NC
GPIO 7->  NC
GPIO 8->  NC
GPIO 9->  NC
GPIO 10-> NC
GPIO 11-> NC
GPIO 12-> Led red
GPIO 13-> NC
GPIO 14-> Led blue
GPIO 15-> NC
GPIO 16-> pin 35 RST CC2652
GPIO 17-> pin 5 CLKIN LAN8720 
GPIO 18-> pin 12 MDIO LAN8720
GPIO 19-> pin 17 TXD0 LAN8720
GPIO 21-> pin 16 TXEN LAN8720
GPIO 22-> pin 18 TXD1 LAN8720
GPIO 23-> pin 13 MDC LAN8720
GPIO 25-> pin 8 RXD0 LAN8720 + MODE0
GPIO 26-> pin 7 RXD1 LAN8720 + MODE1
GPIO 27-> pin 11 CRS_DV LAN8720 + MODE2
GPIO 32-> pin DIO15 FLSR CC2652P
GPIO 33-> pin 2/10 IN1-2/INI3-4 RS2099 (SPDT) 0->Ethernet, 1->USB
GPIO 34-> NC
GPIO 35-> BUTOON config.
GPIO 36-> pin 4 COM2 RS2099 pin (SPDT) 0->GPIO36-DIO13 1->GPIO4-DIO12
GPIO 39-> NC 

I will check in a few days... thanks again for your help.

radioelf, 2024-06-01:
Hi,
I have been able to dedicate some time, and in order to do some testing, I decided to compile the code with a few small changes:

hw.h 
//#define ETH_CFG_CNT 3
//#define ZB_CFG_CNT 8
//#define MIST_CFG_CNT 3
//#define BOARD_CFG_CNT 14
// Mod. for HMG-01 POE Plus device (Chinese clone)
**_#define ETH_CFG_CNT 4 
#define ZB_CFG_CNT 1   
#define MIST_CFG_CNT 2 
#define BOARD_CFG_CNT 13_** 

hw.cpp
EthConfig ethConfigs[] = {
    {.addr = 0, .pwrPin = 12, .mdcPin = 23, .mdiPin = 18, .phyType = ETH_PHY_LAN8720, .clkMode = ETH_CLOCK_GPIO17_OUT, .pwrAltPin = -1}, // 0 Olimex-ESP32-POE
    {.addr = 1, .pwrPin = 16, .mdcPin = 23, .mdiPin = 18, .phyType = ETH_PHY_LAN8720, .clkMode = ETH_CLOCK_GPIO0_IN, .pwrAltPin = -1},   // 1 WT32-ETH01
    {.addr = 0, .pwrPin = -1, .mdcPin = 23, .mdiPin = 18, .phyType = ETH_PHY_LAN8720, .clkMode = ETH_CLOCK_GPIO17_OUT, .pwrAltPin = 5},  // 2 T-Internet-POE
    _**{.addr = 0, .pwrPin = 5, .mdcPin = 23, .mdiPin = 18, .phyType = ETH_PHY_LAN8720, .clkMode = ETH_CLOCK_GPIO17_OUT, .pwrAltPin = 5},   // 3 Mod. for HMG-01 POE Plus device (Chinese clone)**_
};
BrdConfigStruct brdConfigs[] = {
    {"UZG-01", .ethConfigIndex = 2, .zbConfigIndex = 0, .mistConfigIndex = 1},
    {"SLZB-06", .ethConfigIndex = 1, .zbConfigIndex = 1, .mistConfigIndex = 2},
    {"WT32-ETH01", .ethConfigIndex = 1, .zbConfigIndex = 1, .mistConfigIndex = 0},
    {"T-Internet-POE", .ethConfigIndex = 2, .zbConfigIndex = 0, .mistConfigIndex = 0},
    {"Olimex-ESP32-POE", .ethConfigIndex = 0, .zbConfigIndex = 0, .mistConfigIndex = 0},
    {"China-GW", .ethConfigIndex = 0, .zbConfigIndex = 2, .mistConfigIndex = 0},
    {"TubesZB-eth", .ethConfigIndex = 1, .zbConfigIndex = 1, .mistConfigIndex = 0},
    {"TubesZB-eth_usb", .ethConfigIndex = 1, .zbConfigIndex = 4, .mistConfigIndex = 0},
    {"TubesZB-poe", .ethConfigIndex = 0, .zbConfigIndex = 5, .mistConfigIndex = 0},
    {"TubesZB-poe-2022", .ethConfigIndex = 0, .zbConfigIndex = 6, .mistConfigIndex = 0},
    {"TubesZB-poe-2023", .ethConfigIndex = 0, .zbConfigIndex = 7, .mistConfigIndex = 0},
    {"CZC-1.0", .ethConfigIndex = 2, .zbConfigIndex = 0, .mistConfigIndex = 1},
    _**{"HMG-01-POE-Plus", .ethConfigIndex = 3, .zbConfigIndex = 0, .mistConfigIndex = 1},      // 11 Mod. for HMG-01 POE Plus device (Chinese clone)**_
};

With these changes, the system is working correctly (it has been running for ≈45 minutes now... OK), 
but I notice that the local time is not correct; it should be one hour earlier than displayed (daylight saving time in Spain), but this is not an issue for this topic...

xyzroe, 2024-06-01:
Your screenshot shows that your device using UZG-01 config. Not your new one.

xyzroe, 2024-06-01:
Could you please use firmware with no modifications, and found working config just changing configHw.json file

radioelf, 2024-06-02:
Hi,

This morning I found that the equipment was not working; there was no communication with it at all. I couldn't check the status of the LEDs because it is installed and in production in an attic with difficult access. After several power disconnections (POE), I was able to connect and installed the firmware I had previously https://github.com/radioelf/uzg-firmware. This is working without any issues. I don't know the reason for the problem, whether it's hardware or software-related. I will see in a few days if it continues to work without problems to determine if the issue is with my hardware...

radioelf, 2024-06-02:
> Your screenshot shows that your device using UZG-01 config. Not your new one.

It is a mistake I made in the configuration. #define BOARD_CFG_CNT 13

xyzroe, 2024-06-02:
Anyway it will use UZG-01 config. 
If you want to get helped, do what I ask.

radioelf, 2024-06-02:
OK, no problem. I will follow your instructions, but I will wait a couple of days to ensure that my hardware is working correctly. Thanks again for the help.

radioelf, 2024-06-04:
Hi, it seems that the hardware is working correctly since I haven't had any problems these days. Initially, I used the default configuration from configHw.json:
{
  "board": "UZG-01",
  "addr": 0,
  "pwrPin": -1,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": 5,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 33,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}
It seems to work correctly, but sporadically, and often after a reset, it stops working, and it is necessary to disconnect the power supply for it to work correctly. In some cases, it does not start, but if you keep disconnecting and reconnecting the power supply, it eventually works again. For the past few hours, I have been using the following configuration in configHw.json:
 {
  "board": "UZG-01",
  "addr": 0,
  "pwrPin": 5,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": 5,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 33,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}
I only changed 'pwrPin': -1 to 'pwrPin': 5, and it seems that the problem does not occur... I will wait a bit longer and provide an update.

radioelf, 2024-06-05:
Hi, the device has been running without issues for about 24 hours now. I have performed several resets to see if the problem would recur, and it has always initialized without any problems. I believe the modification made is correct for HMG-01 POE Plus. If you need me to perform any additional tests, please let me know ...

xyzroe, 2024-06-05:
So only change 'pwrPin': -1 to 'pwrPin': 5, right?
Could you try to change pwrAltPin to -1?

radioelf, 2024-06-05:
Correct, only 'pwrPin' was changed from -1 to 5, now I have changed 'pwrAltPin' from 5 to -1, and after a reset, it is working without problems...
Current configurationl:
{
  "board": "UZG-01",
  "addr": 0,
  "pwrPin": 5,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "pwrAltPin": -1,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 33,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}

closed 2024-06-06 by xyzroe

w1zz4, 2024-06-16:
@radioelf 

I have the same device and I am not able to put it in flash mode... Did you had to do anything special?

radioelf, 2025-03-12:
> [Ruroelf](https://github.com/radioelf) hola. has instalado beta 20241001 ?? es trabajo? Me sorprende no poder editar configHw.json

Hi, 
No, the beta versions are not installing. I have tried, but after the reboot, it continues with the latest version. I haven't insisted since version 20240914 is working correctly for me.