xyzroe/XZG #54: 503 Error on "General" page when using nginx proxy manager

maxandcheeses, 2024-05-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup reverse proxy that points to XZG device
2. Navigate to <xzg-device>.<your-domain>.com settings
3. Go to 'General' tab

**Expected behavior**
Page should load like all the other tabs

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser Chrome
 - 125.0.6422.77

**Additional context**
All other tabs in the settings page loads just fine

xyzroe, 2024-05-29:
I see 502 error on your screenshot

maxandcheeses, 2024-05-29:
*corrected the mistake

xyzroe, 2024-05-29:
I think the problem is caused by the header response with timezones names. It's large, about 8Kb. Check your nginx proxy for max headers size or so on.

maxandcheeses, 2024-05-30:
You are spot on. I increase the size to address the issue.

```
proxy_buffers   4 32k;
proxy_buffer_size   32k;
```

Thanks for help!

closed 2024-05-30 by maxandcheeses