xyzroe/XZG #56: UZG-01 does not register itself with DNS after the latest version

GD-Dal, 2024-05-30:
UZG-01 does not register itself with DNS after the latest version 20240529.1
It may be introduced in the release before, but it was with this update I discovered it

xyzroe, 2024-06-03:
What did you mean not register itself with DNS?

GD-Dal, 2024-06-03:
Dynamic DNS registration
The old firmwares created a DHCID dns record when supplied with an IP address from DHCP.
The new UXG software does not.

Here is from an UZG running the old firmware:

[image]

From another unit running the UXG firmware:
[image]

GD-Dal, 2024-06-03:
Nevermind this, it seems to work as intended now.

Thanks

closed 2024-06-03 by GD-Dal

xyzroe, 2024-06-03:
The mDNS section in the code for both the XZG and UZG repositories is similar. The primary difference is an additional ServiceTxt entry for "board" in the XZG repository. 
[one](https://github.com/xyzroe/XZG/blob/9859ee85c9d5b17ceea74574696ebee432f0e9ca/src/main.cpp#L405) and [two](https://github.com/mercenaruss/uzg-firmware/blob/813589ad587cbe860589a4334f002e7351525ed1/src/main.cpp#L817)