xyzroe/XZG #57: UZG-01 - Low LQI on all Zigbee devices

GD-Dal, 2024-05-30:
After moving all my devices to the UZG-01, all of them has low LQI.
Some of theme are slow to respond.
The highest connected one has 153 LQI, the lowest one has 29

With my old Conbee II stick, most units had excellent LQI

The Zigbee2MQTT configuration is set to:

# Serial settings
serial:
# Location of XZG
  port: tcp://uzg-01.domain.tld:6638
  baudrate: 115200
# Disable Zigbee led?
	disable_led: false
# Set output power to max 20
advanced:
	transmit_power: 20

xyzroe, 2024-05-30:
This is not related to XZG Firmware. 
But let me remind you that P7 and P2 differ in maximum power, right now. 
Also you should try to use latest beta Zigbee firmware 202403015_202403016

closed 2024-05-30 by xyzroe

GD-Dal, 2024-05-30:
> But let me remind you that P7 and P2 differ in maximum power, right now.

That's news to me.
The web site states:
**POE or USB-C powering, can be powered by type-C power by PoE. PoE+type-C works at the same time as optoelectronic isolation included  CC2652P7 TI chip the most powerful (+20dB), up to 300 Zigbee devices**

Will it be fixed?

Already on the latest zigbee version (202403015_202403016)

xyzroe, 2024-05-30:
Current TI SDK doesn't support 20dB on CC2652P7. 
These question should be addressed TI or Zigbee firmware developer.
XZG is just ESP32 firmware.