xyzroe/XZG #63: Zigbee Firmware value set to "0" after reboot ESP32

reichelper, 2024-05-31:
**Describe the bug**
After reboot the ESP32 with the latest image the Zigbee Firmware Version shows "0"
[See #47](https://github.com/xyzroe/XZG/issues/47)

**Device information**
see Screenshots...

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Reboot ESP32
[image]

2.  Check Zigbee Version
[image]

3. Zigbee Version is shown as expected 
[image]

**Expected behavior**
After Reboot the ESP32 the Zigbee Firmware Version should be show the correct value

xyzroe, 2024-06-08:
@reichelper could you re-flash Zigbee on the latest XZG firmware, even on the same Zigbee version, and then re-check this issue ?

jdm09, 2024-06-09:
Same for me.
Since Update to one of the last 4 new ESP32 Firmwares i get the 0 or ? Firmware Version.
I re-flashed several times.
Only in Console i get the correct Version output.
[image]
[image]

reichelper, 2024-06-09:
@xyzroe 
I flashed the latest ESP32 firmware.
Now after a Reboot of ESP32, the Zigbee Version show "?" instead of "0".

xyzroe, 2024-06-09:
You need to re-flash Zigbee

jdm09, 2024-06-09:
I just tried to flash the Zigbee FW and the UZG-01 hangs up at 72,30 % and the red LED is flashing. Whats this?

xyzroe, 2024-06-09:
So I need to add socket connection and MQTT stoping mechanism. Ok.

closed 2024-06-09 by xyzroe