xyzroe/XZG #70: Socket connection and MQTT need to be stoped while Zigbee flash

xyzroe, 2024-06-09:
So I need to add socket connection and MQTT stoping mechanism. Ok.

_Originally posted by @xyzroe in https://github.com/xyzroe/XZG/issues/63#issuecomment-2156456317_

xyzroe, 2024-06-09:
But serial already didn't work while Zigbee flashing. And MQTT doesn't interfere at all. @jdm09 could you try again but disabling only ZHA ?

jdm09, 2024-06-09:
I tried these steps:
1. ZHA & MQTT enabled and connected -> Coordinator stucks at several progress states (72,3, 83 or 96%). Sometimes the Coordinator restarts and doesnt come back to life. After a power cycle i get the Website again back to life but ZHA Connection always stucks. MQTT Connection was established.
2. ZHA disabled, MQTT enabled -> Update succeeded but the Coordinator restarts after update and wont response. After a power cycle everything works well (HA ZHA Connection & MQTT Connection)
3. ZHA enabled & MQTT disabled -> Same as 1.

So i think may a ZHA or Socket Connection stop mechanism should work properly.

Thanks very much for your support!

xyzroe, 2024-06-09:
Thanks for your feedback

LeonArmston, 2024-06-09:
Yes a handy hint would be to tell people they might want to shutdown their instance of Zigbee2MQTT or ZHA down whilst updating Zigbee. I was only able to update OTA by stopping Zigbee2MQTT

LeonArmston, 2024-06-10:
> It seems to me that if you are going to update the firmware of a device, it would be logical to stop using it while the firmware is being updated.

Indeed - I was putting this message up and suggesting a warning so future me remembers to stop Zigbee2MQTT first (or otherwise it freezes during the flash at x%). 💯

jdm09, 2024-06-10:
Because of the Zigbee FW Bug [koenkk ZG FW Bug on P7 Coordinators #496](https://github.com/Koenkk/Z-Stack-firmware/discussions/496)
i had to update this. Now i had the XZG Fail as discussed above: ZHA was disabled, MQTT enabled and connected. While Update Process, it stopped again at xx % and i had to power cycle to get WebUI back to life. After that i disabled MQTT also and ZB FW Update finished successfully.

Would be great, if you could post a warning note or disable ZHA and MQTT before update starts. Thanks a lot for this support here!

xyzroe, 2024-06-10:
Guys, thanks a lot for your feedback. I'll keep it in mind.

xyzroe, 2024-06-10:
@LeonArmston @jdm09 
try please [this one](https://github.com/xyzroe/XZG/releases/tag/20240610), **without** disabling ZHA and MQTT

Jeppedy, 2024-07-14:
Help me out here...  What's the clean process for upgrading the Zigbee firmware, or the ESP32 firmware?
I was working just fine (though a new user) and upgraded to the 10-July Zigbee firmware and all hell has broken loose.
I have rolled back the update.  I've rebooted everything repeatedly.  Everything seems to be working, I'm getting updates to the MQTT topics from my devices.  But Z2M can't seem to reach any of my devices.  
The hardest part is there's no real debugging available.  How can I be seeing updates in MQTT and yet Z2M says it can't connect to any of my devices.
Any help here?  How am I supposed to properly update the UZG-01?

Jeppedy, 2024-07-14:
> It seems to me that if you are going to update the firmware of a device, it would be logical to stop using it while the firmware is being updated.

I could agree with you, but unless you are deep into Zigbee communications, this is how it appears to people:
So, this device has an update available.
I tell it to update.
I expect the device will do whatever it needs to do to update properly.

I don't usually have to think about what else may be going on when I tell a device to update.
And if I AM supposed to stop the Zigbee communications, how do I disconnect the UZG-01/XZG device from the airwaves and Z2M and still tell it to update?  Why would I need to shut down Z2M?  Can't the UZG-01/XZG just "stop talking", do the upgrade, then start talking again??

xyzroe, 2024-07-14:
> Help me out here... What's the clean process for upgrading the Zigbee firmware, or the ESP32 firmware? I was working just fine (though a new user) and upgraded to the 10-July Zigbee firmware and all [...]

1. Updating any firmware requires stopping the software managing the Zigbee network before the update or restarting it immediately after the update.
2. Your mistake was that you started doing everything at once instead of understanding what was happening and how to proceed correctly.
3. If your devices are in MQTT but not in z2m, then another software is processing the Zigbee network messages. You should not use two software products simultaneously unless you are well-versed in this.
4. You need to debug not the XZG but the software solution that is connected to the gateway. It seems you are using z2m, but likely not only it.
5. It seems to me that you have added your coordinator to ZHA without realizing it.

Jeppedy, 2024-07-14:
>    1. XZG firmware, like UZG gateways and similar devices, are not
>    end-consumer products. Let's call them smart home enthusiast products
>    intended for testing and debugging software solutions. All responsibility
>    lies with you, and understanding the processes is required.
>    2. The device does everything necessary for the update on its own.
>    Your mistake occurred after the update was completed.
>    3. You do not need to disconnect the device for the update, only
>    stop/restart the software you are using.
>    4. After the firmware update, the device is "clean" and needs to be
>    brought back to a working state, which z2m does during its startup process.
>    It is not necessary to stop z2m; the gateway will stop communicating with
>    z2m on its own, but you must restart z2m immediately after the update. Many
>    users prefer to stop z2m before the update and start it afterward to be
>    safe. This is because z2m can make a backup of the coordinator's memory
>    during shutdown (on a clean firmware, this would be an empty backup), but
>    not all software solutions have this behavior.

Thanks for the info.  After I let it just sit overnight, it healed.
So the Z2M component doesn't need to be shut down, as the update process
stops communicating before starting the update.
The help is appreciated!

Rxich, 2026-02-01:
To all who try to update. I was unable to update even with shutting down Z2M & MQTT on HA. I even tried 4 different cables w/ serial->USB, no luck error same as OP. It was only when I shutdown HA completely and used the netwrok connection via the UZG webpage was I able to flash the coordinator  100% without any stalling. Hopefully my HA setup works when I restart