xyzroe/XZG #78: Wireguard no longer connects after upgrade from FW revision 20240519 to 20240612

Oliviakrkk, 2024-06-13:
**Describe the bug**
After upgrading FW revision 20240519 to 20240612 wireguard is no longer connecting. It shows status as Initialized and Disconnected. 

On the affected firmware is not connecting. 
[image]

Log:
```
[1210] | [ZB] Chip: CC2652P7
[1210] | [ZB] IEEE: xxxx
[1221] | [ZB] Flash size: 704 KB
[2347] | [Time] Thu Jun 13 19:50:08 2024
[2348] | [UPD_CHK] cron 9 30 19 * * *
[2453] | WG was initialized
[5689] | [DNS] Saved WiFi - 192.168.101.44
[14985] | [ZB] FW: Unknown! Check serial speed!
```

After downgrade:
[image]

Log:
```
[131] | Switched USB to ESP
[1210] | [ZB] Chip: CC2652P7
[1210] | [ZB] IEEE: xxxx
[1220] | [ZB] Flash size: 704 KB
[3053] | [ZB] fw: unknown!
[3997] | WG was initialized
[4009] | Local time: Thu Jun 13 19:39:16 2024
```

**Device information**
- Screenshot of Zigbee and Device tab from root page of device
- JSON object returned on [http://xzg.local/api?action=1&param=all](http://xzg.local/api?action=1&param=all) 
 *You **should** remove your credentials first*
  
[image]

```
{
  "network-config": {
    "wifiEnbl": true,
    "wifiSsid": “xxxxxx”,
    "wifiPass": “xxxxxxx”,
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": false,
    "ethDhcp": true,
    "ethIp": "::",
    "ethMask": "255.255.255.0",
    "ethGate": "::",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": true,
    "wgLocalIP": “xxxxx”,
    "wgLocalSubnet": "255.255.255.255",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": “xxxx+xxxxx=",
    "wgEndAddr": “xxxxx”,
    "wgEndPubKey": “xxxxx”,
    "wgEndPort": 13231,
    "wgAllowedIP": “xxxxxx”,
    "wgAllowedMask": "255.255.255.255",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-CD6A",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": false,
    "server": "",
    "port": 1883,
    "user": "",
    "pass": "",
    "topic": "XZG-CD6A",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 15
  },
  "system-config": {
    "disableWeb": true,
    "webAuth": true,
    "webUser": “xxxx”,
    "webPass": “xxxxx”,
    "fwEnabled": true,
    "fwIp": “xxxx”,
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 15,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 1,
    "hostname": "UZG01-xxxx”,
    "timeZone": "Europe/xxxx”,
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 0,
    "zbFw": "?",
    "updHour": "19:30",
    "updDays": "*",
    "autoIns": true
  }
}
```

**To Reproduce**
Upgrade to 20240612

**Expected behavior**
Wireguard client should connect.

xyzroe, 2024-06-13:
wgEndAddr is IP or hostname ? Could you try using IP ?

Oliviakrkk, 2024-06-13:
It is an IP.

Oliviakrkk, 2024-06-14:
It looks like it never worked after upgrazing to XZG. 

Looking at this function: https://github.com/xyzroe/XZG/blob/main/src/etc.cpp#L828

The configuration should be OK. 

Comparing to the previous config you just give more options when initializing: https://github.com/mercenaruss/uzg-firmware/blob/main/src/main.cpp#L91C6-L91C13

Oliviakrkk, 2024-06-14:
@xyzroe 
In this example: https://github.com/ciniml/WireGuard-ESP32-Arduino/blob/main/examples/uptime_post/uptime_post.ino
there are only 5 variables for `wg.Begin`. Maybe the extra config is creating issues?

xyzroe, 2024-06-14:
> It looks like it never worked after upgrading to XZG. 
> 
no you're wrong. I made some tests and get stable working connection

xyzroe, 2024-06-14:
> Comparing to the previous config you just give more options when initializing: 

yep, cause of using another version of WireGuard library

xyzroe, 2024-06-14:
https://github.com/Tinkerforge/WireGuard-ESP32-Arduino

Oliviakrkk, 2024-06-14:
The test you mention, which version did you test?

xyzroe, 2024-06-14:
Need to check when additional config options added. That way we can found tested version

Oliviakrkk, 2024-06-24:
Hi, 
Maybe I am configuring it wrongly. 

In the previous version of the library, there wasn't a variable called `Local Gateway`. I managed to connect (once, somehow) when this was set to nothing, but there was no comunnication via the wireguard interface. 

What should I enter into this field? The gateway of the local network the device is connected to? 

I tried to find the variable meaning in repo https://github.com/Tinkerforge/WireGuard-ESP32-Arduino, but I just can see the basic vars which were used in the UZG.

xyzroe, 2024-06-24:
Have you tried with default Local gateway = 0.0.0.0 ?

Oliviakrkk, 2024-06-24:
Yes. "Worked" once. Handshake with my router was completed, but non of the communucation worked. Ping did not work. 

Now I am trying to manually enter the gateway from the wifi network that the UZG is connected to.

Oliviakrkk, 2024-06-24:
BTW. There are two more commits in the library after linked commit `[WireGuard @ 58de5da](https://github.com/Tinkerforge/WireGuard-ESP32-Arduino/tree/58de5da725bf11720439306496d8962dcb26af61)`

https://github.com/Tinkerforge/WireGuard-ESP32-Arduino/activity

I am honestly not that much fluent to be able to asses if they would make the difference...

xyzroe, 2024-06-24:
Ok, I'll look in a few days, thanks.

gztproject, 2024-06-26:
It works for me in version 20240612:

"vpn-config": {
    "wgEnable": true,
    "wgLocalIP": “10.100.0.4”, //Peer address
    "wgLocalSubnet": "255.255.255.0", //Peer netmask
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": “xxxx+xxxxx=",
    "wgEndAddr": “xxxxx”, //Server address
    "wgEndPubKey": “xxxxx”,
    "wgEndPort": 51820,
    "wgAllowedIP": “0.0.0.0”, //What IPs are allowed to connect to this peer
    "wgAllowedMask": "0.0.0.0", //Subnet for ^
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
  },

gztproject, 2024-06-28:
I just opened a pull request #86 with a quick VPN runthrough :)

closed 2024-07-07 by xyzroe