xyzroe/XZG #81: SLZB-06 - No network, no ZIGBEE

w1zz4, 2024-06-18:
**Describe the bug**
NEW SLZB-06, working fine on SMLIght Firmware, no network and Zigbee on XZG.
Board and zigbee are not detected properly (Unknown). I suspect a new rev with maybe a different pinout or something but can't fin info on SMLight website.

**Device information**
SMLIGHT:[image]
XZG:[image]
http://ZXG.local/api?action=1&param=all :
{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": false,
    "ethDhcp": true,
    "ethIp": "::",
    "ethMask": "255.255.255.0",
    "ethGate": "::",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "::",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-ACAE",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": false,
    "server": "",
    "port": 1883,
    "user": "",
    "pass": "",
    "topic": "XZG-ACAE",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 15
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": false,
    "webUser": "",
    "webPass": "",
    "fwEnabled": false,
    "fwIp": "0.0.0.0",
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 9,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 1,
    "hostname": "XZG",
    "timeZone": "Europe/Berlin",
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 0,
    "zbFw": "?",
    "updHour": "01:00",
    "updDays": "*",
    "autoIns": false
  }
}

Logs : 
[12072] | [ZB] FW: Unknown! Check serial speed!
[300488] | [Time] Thu Jan  1 01:05:00 1970
[300489] | [UPD_CHK] cron 40 0 1 * * *
[313227] | [Time] Thu Jan  1 01:05:13 1970
[313228] | [UPD_CHK] cron 6 0 1 * * *
[358552] | [ZB] No Zigbee chip!  

**To Reproduce**
Install either 20240612 or 20240609.01 firmware

**Expected behavior**
Network and Zigbee working fine

**Additional context**
I try 20240612 and 20240609.01. Also I did an ESP32 and reset and NVRAM erase without success.

xyzroe, 2024-06-18:
Could you try to edit configHw.json in Tools-File browser 
to make it looks like this
```
{
  "board": "SMZB-06",
  "addr": 1,
  "pwrPin": 16,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 1,
  "clkMode": 3,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 4,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 17,
  "zbRxPin": 5,
  "zbRstPin": 33,
  "zbBslPin": 32
}
```

Then restart the device and make video how LEDs will blink. If device won't start you need to upload that video somewhere, or just describe how it was.

w1zz4, 2024-06-18:
Gonna try tomorrow and take a video of the LEDs behaviour

w1zz4, 2024-06-19:
Ok, now with that config, Zigbee is correctly detected and I was able to flash it.
Unfortunately still no eth network. Wifi is working correctly.

If it can help, this time (for the first time). Eth Network was working when I flashed and I received a popup asking me to choose my board between WT32-ETH01 and TubesZB-eth. Then when I change configHW.json for the one you gave me, eth stopped working. Unfortunetly, I did not take a copy of configHW.json when it was working and I was not able to get that behaviour since...

Here is current status :
[image]

xyzroe, 2024-06-19:
You can try to make a factory reset, via Tools page. It will ask you to choose the board again, after restart. Just to check configHw.json for correct ETH section.

xyzroe, 2024-06-19:
Have you looked in the Network page for the Eth config section after applying new configuration

w1zz4, 2024-06-19:
I did a reset and after reboot went back to Unknown Model and Ethernet not working. reflash smlight then flashback xzg and still the same.

when I enable Ethernet from network setting it’s stays disconnected, Mac is all 0s and I all configuration field are grayed/locked.

xyzroe, 2024-06-19:
I have no ideas. Maybe you should try to ask here https://youtu.be/24C8liZYGWY?feature=shared&t=549 . Some guy got SLZB-06 works on XZG

w1zz4, 2024-06-20:
I wil check this out and if I find any solution will open new issue

erikpendragon, 2024-06-30:
@w1zz4  I got it working with this config file. If you are flashing with the online web flasher and you are on a Mac. It gives 2 usb serial devices. try the other one, when I did it auto detected everything and this was the config file it had in it.
{
  "board": "SLZB-06",
  "addr": 1,
  "pwrPin": 16,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 0,
  "pwrAltPin": -1,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 4,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 17,
  "zbRxPin": 5,
  "zbRstPin": 33,
  "zbBslPin": 32
}

w1zz4, 2024-07-03:
> @w1zz4 I got it working with this config file. If you are flashing with the online web flasher and you are on a Mac. It gives 2 usb serial devices. try the other one, when I did it auto detected everything and this was the config file it had in it. When using the web flasher I also tried an old version 20240507, don't know if that helped. 

I flashed with 20240507 this morning and it detected the board correctly which confirm a bug in newer firmwares.

Thank You

w1zz4, 2024-07-03:
> If anyone can help me with debugging some test builds, I can try to find and fix that bug.

I am open to do it

erikpendragon, 2024-07-03:
I can, but my SLZB-06 is not fully functional. Something with the process my Ethernet port doesn't work anymore even with the settings I had before that did work. 
Was talking with SMLight and they said the port is dead.

w1zz4, 2024-07-03:
After first successful installation, I can install latest firmware for both Zigbee radio and ESP without issue.

Here is my current configHw.json
```
{
  "board": "SLZB-06",
  "addr": 1,
  "pwrPin": 16,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 0,
  "pwrAltPin": -1,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 4,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 17,
  "zbRxPin": 5,
  "zbRstPin": 33,
  "zbBslPin": 32
}
```

erikpendragon, 2024-07-03:
@w1zz4 does the Ethernet now work, because mine says enabled but disconnected like your pictures above. Although it did work at one point.

w1zz4, 2024-07-03:
Yes everything is up and running.

[image]

erikpendragon, 2024-07-03:
Well my zigbee network is down anyways, might as well try some custom builds.

xyzroe, 2024-07-04:
> they said the port is dead.

How did they explain why the port is dead?

xyzroe, 2024-07-04:
> Well my zigbee network is down anyways, might as well try some custom builds.

What you are going to achieve using custom builds? You could change entire hardware config using configHw.json file.

xyzroe, 2024-07-07:
Could someone make tests with SLZB-06 and confirm that board detection works right?

w1zz4, 2024-07-07:
Tomorrow I should be able to test it

w1zz4, 2024-07-08:
Unfortunately, after factory reset board is again not detect correctly.

Copying working json over new one did not make it works either, Had to reinstall 20240507 then do another factory reset to have my board detected properly again.

er1z, 2024-10-29:
Any news on this?

basalisco, 2025-01-09:
same problem ethernet not recognized.

xyzroe, 2025-01-09:
Looks like it can be fixed in a new version (still in work, no ETA)

parsley, 2025-07-28:
Any news on this?

I got my first SLZB-06 and it looks like I ran into the same issue:
Flashing any version from https://smlight.tech/flasher/#SLZB-06 (currently v2.9.4 or v2.7.1 or v2.3.6 or v0.9.9 are offered directly) works fine.
Installing any recent XZG version like 20240914 or 20240928 or 20241001 via web-flasher (brave browser on a Mac) results in strange behavior like "unknown hardware", no eth and so on. 
Not mentioned so far: When I open tools-debug-file browser there is no configHw.json but some small 7 byte file...
To me this looks like current versions fail to setup a working LittleFS or something similar.

Installing some older version of XZG first I find a valid configHw.json and the device seams to work fine. I can even update to version 20240914 via the tools-firmware-esp32-show available. But uodating beyond that didn't work.

I hope I was able to provide additional hints, even though I have no idea what's going on...

BTW: The slikscreen on the PCB in my SLZB-06 states it's r1.70.

xyzroe, 2025-07-29:
As you can see, there haven't been any commits or releases recently. I don't have enough time or motivation for this project right now. No one else does either. So, the project is on hold for the time being.

ms1958, 2025-11-04:
wie rufe ich die  configHw.json  auf ?
beziehungsweise, wie benutze ich bei Tools/Debug den filebrowser ?
das feld ist bei mir leer ......

How do I access the configHw.json file?
Or rather, how do I use the file browser in Tools/Debug?
That field is empty for me...

LG ms1958

xyzroe, 2026-11-15:
Could you please try the new pre-release [20261115](https://github.com/xyzroe/XZG/releases/tag/20261115)? If the board is still not detected, please post the `param=all` output and the serial log from boot (115200 baud).

ms1958, 2026-11-15:
[image]

Thanks for the new Firmware
regards ms1958