xyzroe/XZG #82: The Delete-button in File Browser is to easy to misinterpret

flemmingss, 2024-06-26:
This delete button looks like magic wand to me.
And on top of that, it is not confirming deletion
[image]

Request:

1. A better explanatory icon, and maybe a text
2. A popup to confirm deletion

zappydood, 2024-06-30:
This just took down my entire zigbee network and I have no idea what is wrong with my slzb-06. It wont open the web panel, and I am having an extremely hard time finding out anything about anything. WHAT did I accidentally do!?

flemmingss, 2024-06-30:
> This just took down my entire zigbee network and I have no idea what is wrong with my slzb-06. It wont open the web panel, and I am having an extremely hard time finding out anything about anything. WHAT did I accidentally do!?

Try a reboot of coordinator, worked for me.

commit 7d066aa3 (2024-07-07): 20240707

- Changed file delete icon - fix #82
- Modify some Chinese words #84 - Thanks @yzwudong 🏆
- Update ru.json #79 - Thanks @valery7373 🏆
closed 2024-07-07 by xyzroe