xyzroe/XZG #88: Unexpected reboot of esp32

yzwudong, 2024-07-10:
**Describe the bug**
Sometimes the device auto reboot.
log is here：
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12736
load:0x40080400,len:2716
entry 0x40080594
[35m1146 [0m[32m[printLogMsg] [0m[ZB] Chip: CC2652P7
[35m1147 [0m[32m[printLogMsg] [0m[ZB] IEEE: 00:12:4B:00:2A:7B:4D:01
[35m1147 [0m[32m[zbHwCheck] [0mmodeCfg ffc1b9fd
[35m1157 [0m[32m[zbHwCheck] [0mbslCfg c50ffec5
[35m1158 [0m[32m[printLogMsg] [0m[ZB] Flash size: 704 KB
[35m1724 [0m[32m[setup] [0mConfig:
{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": true,
    "ethDhcp": true,
    "ethIp": "::",
    "ethMask": "255.255.255.0",
    "ethGate": "::",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "::",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-B196",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": false,
    "server": "",
    "port": 1883,
    "user": "",
    "pass": "",
    "topic": "XZG-B196",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 15
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": false,
    "webUser": "",
    "webPass": "",
    "fwEnabled": false,
    "fwIp": "0.0.0.0",
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 13,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 1,
    "hostname": "XZG",
    "timeZone": "Europe/Berlin",
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 0,
    "zbFw": "?",
    "updHour": "01:00",
    "updDays": "*",
    "autoIns": false
  }
}
[35m1880 [0m[32m[setup] [0mVARS:
{
  "system-vars": {
    "hwBtnIs": true,
    "hwLedUsbIs": true,
    "hwLedPwrIs": true,
    "hwZigbeeIs": true,
    "connectedClients": 0,
    "socketTime": 0,
    "connectedEther": false,
    "apStarted": false,
    "wifiWebSetupInProgress": false,
    "vpnWgInit": false,
    "vpnWgConnect": false,
    "vpnWgPeerIp": "::",
    "vpnWgCheckTimeKey": 0,
    "vpnHnInit": false,
    "mqttConn": false,
    "mqttReconnectTime": 0,
    "mqttHeartbeatTime": 0,
    "disableLeds": false,
    "zbFlashing": false,
    "deviceId": "XZG-B196",
    "espUpdAvail": false,
    "zbUpdAvail": false
  }
}
[35m1942 [0m[32m[setupCoordinatorMode] [0mNetwork
[35m3628 [0m[32m[mDNS_start] [0mmDNS responder started on XZG.local
[35m3642 [0m[32m[printNVSFreeSpace] [0mTotal: 630, Used: 111, Free: 519
[35m4473 [0m[32m[NetworkEvent] [0mETH MAC: 10:06:1C:BB:82:17, IP: 192.168.0.46, Mask: 255.255.255.0, Gw: 192.168.0.1, DNS: 192.168.0.1, 100Mbps
[35m4485 [0m[32m[printLogMsg] [0m[DNS] Saved ETH - 192.168.0.1
[35m5624 [0m[32m[printLogMsg] [0m[Time] Wed Jul 10 15:24:44 2024
[35m5624 [0m[32m[printLogMsg] [0m[UPD_CHK] cron 43 0 1 * * *
[35m15413 [0m[32m[printLogMsg] [0m[ZB] FW: Unknown! Check serial speed!
[35m15413 [0m[32m[setup] [0m[ESP] FW: 20240710
[35m15413 [0m[32m[setup] [0mLoad cfg ZBC03P
[35m16176 [0m[32m[setup] [0mdone
**Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (update Web Task) 
Core  0 register dump:
PC      : 0x40084cd5  PS      : 0x00060736  A0      : 0x3ffda3a0  A1      : 0x3ffda2e0  
A2      : 0x00000000  A3      : 0x00060723  A4      : 0x00060720  A5      : 0x3ffcc370  
A6      : 0x3ffdae3c  A7      : 0x00000002  A8      : 0x3ffc8514  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00060723  A12     : 0x00000001  A13     : 0x3ffda4af  
A14     : 0x3ffb7fc8  A15     : 0x3ffbc9f0  SAR     : 0x0000001c  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008a24d  LEND    : 0x4008a25d  LCOUNT  : 0xfffffffb  

Backtrace: 0x40084cd2:0x3ffda2e0 0x3ffda39d:0x3ffda3b0 |<-CORRUPTED
ELF file SHA256: 0000000000000000

Rebooting...**
ets Jul 29 2019 12:21:46

**Device information**
{
  "board": "ZBC03P",
  "addr": 1,
  "pwrPin": -1,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "btnPin": 34,
  "btnPlr": 1,
  "uartSelPin": 13,
  "uartSelPlr": 1,
  "ledModePin": 4,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 5,
  "zbRxPin": 16,
  "zbRstPin": 33,
  "zbBslPin": 32
}

xyzroe, 2024-07-15:
What board you're using?

yzwudong, 2024-07-21:
////
"workMode": 0,
"zbRole": 0,
"zbFw": "?",    
/////
This issue occurs after a failed role switch

xyzroe, 2024-07-21:
ESP won't reboot if it couldn't detect Zigbee version. But I asked you some question. If you need help - provide information.

xyzroe, 2024-07-27:
@yzwudong will you provide any additional information?

codinv, 2024-09-01:
i have some trouble
device HMG-01 Plus
board HG POE Plus
[image]

xyzroe, 2024-09-14:
Try to disable cron based update availability check

ffries, 2025-04-25:
Same issue here, reboots several times a day:
UZG01: ESP32-D0WDQ6 (dual core)
Zigbee: 20240710
XZG: 20241001

ffries, 2025-04-25:
Hi there,

First many thanks for contributing this very nice piece of software. Same issue here, several reboots a day until I found this bug issue. The settings in the web interface were hard to find. Had to set "When to check" to "Never". Maybe it should be "Check for new firmware" => "Disable".

Kind regards,

ffries, 2025-07-15:
Hello,

With a static IP and MQTT disabled I have an uptime around a day or so. I becomes difficult to monitor downtimes/uptimes because MQTT is disabled, so I enabled MQTT back. The UZG01 reboots so quickly that it is very difficult to notice any downtime. But it is still a problem to know that at some point the device reboots. It looks like a deeply burried issue.

Kind regards,