xyzroe/XZG #91: Ita translate

kirkoff71, 2024-07-14:
Are present error italin translate...
If do you want I can contribute.
Thanks

kirkoff71, 2024-07-14:
Found and correct the "it.json"

xyzroe, 2024-07-15:
Please make a PR with your contribution.

kirkoff71, 2024-07-15:
Just made yesterday.
🙂

xyzroe, 2024-07-15:
> Just made yesterday. 🙂

No. Please check twice. Every Italian will say Grazie to you.
[image]
[image]

xyzroe, 2024-07-15:
I found you repository. You made new branch - super. Now you need to make pull request to push changes from your repository to this one.

kirkoff71, 2024-07-16:
Now I make... check if is ok.
Thanks for your work :)

xyzroe, 2024-07-16:
I still don't have any PRs

xyzroe, 2024-07-16:
You have merged branch with main one but only in you repo. But you need just to open new pull request from you branch patch-1 to this repository main branch. If you open this repo main page in your browser GitHub will offer you to make a contribution using pull request.

kirkoff71, 2024-07-16:
Now?
I attachment the file in this chat now
[it.json](https://github.com/user-attachments/files/16247318/it.json)

xyzroe, 2024-07-16:
when you create a PR it will be visible not only to me or you, but to everyone. I believe you can do it.

closed 2024-07-16 by xyzroe