xyzroe/XZG #92: uzg-01 zigstar wireguard not working

toloomierz, 2024-07-15:
**Describe the bug**
wireguard not working over wifi 

**Device information**
Model	UZG-01
Firmware	20240707
Hardware	ESP32-D0WDQ6

Role	Coordinator
Firmware	20240710

xyzroe, 2024-07-15:
Have you check that config over LAN ?

toloomierz, 2024-07-15:
No only via wifi , and i watched tcpdump and there  was no traffic at all

xyzroe, 2024-07-15:
Please make one more test but using wired connection to check your configuration.

toloomierz, 2024-07-16:
I'll let U know

toloomierz, 2024-07-16:
It's working via ethernet but wifi alone not

toloomierz, 2024-07-17:
it works when i connect to vpn via eth and wifi on and  when i disconnect eth, connection over wifi  is still enabled and working , but wifi mode is changing from client to AP, as below 

 "Mode	AP
SSID	XZG-B1BB
DHCP	Enabled
IP	[172.16.2.111](http://172.16.2.111/)
Subnet	255.255.255.0 (Access point)
Gateway	192.168.1.1 (this device)
DNS	[172.16.1.101](http://172.16.1.101/)
RSSI	
noConn dBm
MAC	94:B9:7E:E2:27:9C"
when i reboot the device, the problem is back and no traffic is going through the wifi interface

xyzroe, 2024-07-17:
Ok. Thanks for investigation. I'll try to fix it and let you know

Oliviakrkk, 2024-07-27:
I guess this is the bug I was hitting when I reported #78 . I always tested using WIFI. 

I have the same problem.

Oliviakrkk, 2024-07-27:
Is there a way to enable debug logs, check routing table or just check the initalization order? I wanted to check if wifi is connected first, then the wg.begin is called.

Oliviakrkk, 2024-07-31:
Hi @xyzroe , 

Is there any test which I could do to help you solve this problem?

xyzroe, 2024-07-31:
No, thanks. 
I need just some free time and motivation 😀
Someday it will happen.

Oliviakrkk, 2024-08-02:
Today I did rollback to reenable zigbee network, as I really need my humidity monitoring back :-).  The new guide how to configure it helped. 

Looks like the last working version is `20240531` and it stopped working in `20240602`. 

https://github.com/xyzroe/XZG/compare/20240531...20240602

toloomierz, 2024-08-02:
@Oliviakrkk , thx for info, I'll do the rollback to 20240531

toloomierz, 2024-10-18:
is this bug solved yet ???

closed 2024-10-19 by toloomierz

Oliviakrkk, 2024-11-11:
@toloomierz on which firmware? I installed `20240914` and are unable to connect.