xyzroe/XZG #98: Support for capturing traffic to/from a Zigbee device?

victorhooi, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
I'm trying to diagnose an issue with a Zigbee device and zigbee2mqtt (https://github.com/Koenkk/zigbee2mqtt/issues/23204 - to be specific)

**Describe the solution you'd like**
There's some discussion here about some ways of sniffing Zigbee traffic:

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#_3-sniffing-traffic

However, I'm wondering if there's any way for the XZG firmware itself to capture/dump the Zigbee traffic, either for everything, or for a specific device (more helpful) - either to local storage, or perhaps to a network volume, or something else like that?

**Describe alternatives you've considered**
Using a second adapter that's close by to try to sniff the traffic (although worried it may miss some).

**Additional context**
Add any other context or screenshots about the feature request here.

xyzroe, 2024-07-26:
Sorry but it is impossible. 
## **The Need for a Second Adapter** 

###  Complete Traffic Capture:

Radio Channel Monitoring: The primary reason for using a second adapter is to monitor the actual traffic on the radio channel. The XZG firmware will only process and display the traffic it successfully receives and transmits. However, not all packets may reach or be received by the XZG due to various reasons like interference, signal loss, or hardware limitations.
Missed Packets: Sniffing with a second adapter helps identify packets that fail to reach the XZG, providing a complete picture of the network's communication, including any lost or corrupted packets.
Debugging and Diagnostics:

Unsuccessful Transmissions: To diagnose issues in the Zigbee network effectively, it is critical to see both successful and unsuccessful transmissions. The XZG firmware can only show packets it handles correctly, missing out on those that were attempted but failed.
Interference and Noise: The second adapter can help detect interference or noise that disrupts communication, which the XZG might not log or report accurately.
Why XZG Firmware Alone Isn't Enough

## Filtered Data:

Processed Packets: The XZG firmware focuses on processed packets, meaning any packet it does not process (due to errors or other reasons) won't be captured or logged. This limitation makes it impossible to see the complete traffic picture using XZG alone.
Visibility of the Entire Network:

End-to-End Monitoring: Capturing traffic at the radio channel level allows for end-to-end monitoring of the entire network. This level of visibility is essential for diagnosing issues that occur at any point in the communication path, not just at the endpoints where XZG operates.

closed 2024-07-26 by xyzroe

victorhooi, 2024-07-27:
@xyzroe Thank you for the detailed explanation - much appreciated!