xyzroe/XZG #99: UZG-01 running 20240707 crashes (stalls) after 1-2 days

victorhooi, 2024-07-27:
**Describe the bug**

I have flashed XYZ onto a UZG-01 (ZigStar UZG) device. It is running as a Zigbee Coordinator in my network, and is connected to Zigbee2MQTT.

It is being powered by PoE.

However, after approximately 1-2 days, the device appears to disappear off the local network - it is no longer responding to ICMP ping, and the web interface is not accessible either, and Zigbee2MQTT is not able to connect to it.

However, the LAN Ethernet activity lights are still flashing, and the left-most blue activity light is flashing. (I can attach a video if needed - or if the frequency etc is somehow relevant).

**Device information**
- Screenshot of Zigbee and Device tab from root page of device
- JSON object returned on [http://xzg.local/api?action=1&param=all](http://xzg.local/api?action=1&param=all) 
 *You **should** remove your credentials first*

Unfortunately I can't get the above, as the device's web interface isn't responding.

Is there some way to get debug logs from the device, before I power-cycle or reboot it? Or what else can I retrieve before I try to reflash it again?
  
**To Reproduce**
Steps to reproduce the behavior:
1. Get a UZG-01.
2. Flash `[20240707](https://github.com/xyzroe/XZG/releases/tag/20240707)` onto the device.
3. Configure it as a Zigbee coordinator, and point a Zigbee2MQTT instance at it.
4. Wait...profit.

**Expected behavior**
The UZG-01 running XZG firmware should not crash or become unavailable.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

xyzroe, 2024-07-27:
I'm also using UZG with 20240707 and don't have such problems.
We need to find something special in your setup.

victorhooi, 2024-07-27:
Got it - is there some way of retrieving any debug data from the running unit (e.g. via the USB port)?

I haven't power-cycled the unit yet, as it's currently in the crashed (wedged) state - I was waiting to check if there might be useful diagnostic information to pull?

xyzroe, 2024-07-29:
You're ok. 
you can build the debug version of firmware it includes more logs.

victorhooi, 2024-07-29:
Got it - is there a place I can download the debug build already for [20240707](https://github.com/xyzroe/XZG/releases/tag/20240707)?

Or do you happen to know if this will build easily on macOS? (Otherwise I can see about setting up a Linux VM to do this all on. Sorry if this is a silly question - but should I assume it will cross-compile to ARM fine, from a x64 box?)

Also - how do I enable the debug build? And where will the additional debug information get dumped to afterwards? Or how do you view these logs?

xyzroe, 2024-07-30:
I'm using mac os. Just install all dependencies

xyzroe, 2024-07-30:
Just select "debug-solo" build variant

MaartenMol, 2024-08-28:
Any update on this?

cshuttle, 2024-09-03:
I am having this issue also, resulting in an unavailable controller several times a day.

sneak-o-matic, 2024-09-07:
for me the same

airecken, 2024-09-10:
I am having the same issue.

Device: Zig-Star UZG-01
Hardware: CC2652P7
Firmware: 20240710

Setup as coordinator.
Happens with fixed IP or DHCP, powered via USB-C ror POE.

xyzroe, 2024-09-14:
Try to disable cron based update availability check

airecken, 2024-09-14:
> Try to disable cron based update availability check

I updated to 20240914, set frequency to disable.  Still automatically reboots itself usually within 15 minutes.  

After multiple self reboots the device usually hangs and requires a power cycle.

MaartenMol, 2024-09-24:
Any update on this?

spitfire, 2024-10-10:
Doesn't happen to me, I'm running a newer version of Zigbee firmware (20240710) though. Have you tried upgrading it?
[image]

shyawnkarim, 2024-10-17:
This same thing has been happening to me for months. I have one connected via PoE with a static IP and DHCP disabled. Its connected to HA via Z2M and 1 or 2 times a day HA will lose connection with the ZigStar and the zigstar will also become unreachable. I end up unplugging it, plugging it back in, and then it connects and behaves for a bit till i have to restart it again.

sneak-o-matic, 2024-10-21:
Now i bumped to latest pre-release (20241001) and then updated zigbee firmware to latest (20241001) and disabled firmware checks. Its running stable since three days. Gonna update if that changes.

airecken, 2024-10-22:
I installed latest pre-release (20241001) with the newest Zigbee firmware I could find (20240710) and that seems to be stable so far as well.  It's definitely lasted longer than before without crashing, so I think that memory leak maybe fixed.

Curious which 20241001 Zigbee firmware you were installing?

sneak-o-matic, 2024-10-22:
@airecken i am sorry. that was a mistake by me. you are right latest zigbee is 20240710. i correct that.

aaronbrooks-gh, 2025-06-21:
I'm experiencing this on UZG-01 with device firmware 20241001 and zigbee firmware 20240710 :(

hexajon-art, 2025-09-24:
Hello! I'm experiencing the same issue as well, even on the same firmware versions that seem to have worked for some people.

After a couple of hours the Web GUI becomes unavailable. The device stays reachable via ping, ZHA connection also keeps working until I restart Home Assistant. Then I have to reboot the device by unplugging the cable and plugging it back in, otherwise Home Assistant can't reconnect.

```
http http://10.10.10.34

http: error: ConnectionError: ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer')) while doing a GET request to URL: http://10.10.10.34/
```

- Device: UZG-01 with CC2652P7
- ESP Firmware: 20241001
- Zigbee Firmware: 20240710
- Powered via PoE
- Connected via Home Assistant ZHA serial socket

As the issue is still open, I'm assuming the issue still persists. Any update or ideas?

hexajon-art, 2025-11-21:
@xyzroe Thank you for your response.

- I've tried multiple versions in the last months and on 0914 the same thing happens.
- The ZHA connection via socket used to work, but the connection to the webserver on the ESP just breaks. I can't access the XZG web UI and if I restart my Home Assistant, it can't reconnect to the socket until I also restart the device. For the last few days I can't get a connection at all anymore, my smart home is pretty dumb right now
- I'm powering the device via PoE, I can try USB-C. I will post photos once I get a chance to open the device.

hexajon-art, 2025-11-28:
Hello @xyzroe 

Here is my current setup:

[image]

* Device: UZG-01 with CC2652P7
* Erased NVRAM via XZG UI
* Factory reset via hardware button on device
* ESP Firmware: 20240914 (Flashed via XZG MT)
* Zigbee Firmware: 20250321 (Flashed via XZG UI > Zigbee > Device Role > Coordinator)
* Powered via USB-C
* Ethernet cable plugged in on a non-PoE port on my switch
* Both cables are long and I've moved the device 3m away from any computer to minimize interference
* DHCP disabled, static IP
* Firmware update check disabled
* MQTT disabled
* I've deleted ZHA from my Home Assistant installation
* ZHA setup fails when trying to create a network
* Here is a photo of the inside:

[image]

Home Assistant log from failed ZHA setup:
```python
2025-11-28 16:27:59.827 ERROR (MainThread) [aiohttp.server] Error handling request from 10.10.10.21
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/api.py", line 1079, in request
    response = await response_future
               ^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 207, in create_zigpy_app
    await app.connect()
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/zigbee/application.py", line 73, in connect
    await znp.connect()
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/api.py", line 752, in connect
    self.version = await self.detect_zstack_version()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/api.py", line 93, in detect_zstack_version
    await self.nvram.read(
    ...<3 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/nvram.py", line 329, in read
    read_rsp = await self.znp.request(
               ^^^^^^^^^^^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/api.py", line 1075, in request
    async with asyncio_timeout(
               ~~~~~~~~~~~~~~~^
        timeout or self._znp_config[conf.CONF_SREQ_TIMEOUT]
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ):
    ^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
[... 108 lines cut ...]
    await self.start_network()
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/zigbee/application.py", line 151, in start_network
    await self._write_stack_settings()
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/zigbee/application.py", line 642, in _write_stack_settings
    current_value = await self._znp.nvram.osal_read(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        nvid, item_type=type(value)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/nvram.py", line 168, in osal_read
    length = (await self.znp.request(c.SYS.OSALNVLength.Req(Id=nv_id))).ItemLen
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy_znp/api.py", line 1075, in request
    async with asyncio_timeout(
               ~~~~~~~~~~~~~~~^
        timeout or self._znp_config[conf.CONF_SREQ_TIMEOUT]
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ):
    ^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
``` 
Seems like the ZHA integration isn't getting any response when trying to set up the network. Any idea what might be going on?

xyzroe, 2025-11-28:
1. Your device looks like an original ZigStar UZG-01, so I don't think there's a hardware issue.

2. it doesn't looks like ZHA integration isn't getting any response 
> 2025-11-28 16:43:15.962 INFO (MainThread) [zigpy.device] [0x0000] Discovered basic device information for ZNPCoordinator model='Coordinator' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:2e:0c:c6:9d is_initialized=True

3. Are you trying to set up a new network or use an existing configuration?

4. You can check the Tools > Debug > Console tab in the XZG UI to confirm that the device is receiving and sending data to the ZHA.

5. I tried setting up ZHA in fully automatic mode, but I got an error. Then I tried manual mode, but I got an error again. After that, I started zigbee2mqtt, and there weren't any errors. Everything worked smoothly. So I thinks there's some problem with ZHA.

xyzroe, 2025-11-28:
Sorry, but I've run out of ideas.

hexajon-art, 2025-11-30:
* I've set up an MQTT server
* I've set up Zigbee2MQTT
* When trying to start Z2M it fails with:

```
[2025-11-30 15:14:42] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-11-30 15:14:42] debug:    zh:zstack:znp: --> SREQ: SYS - nvRead - {"sysid":1,"itemid":1,"subid":97,"offset":0,"len":12}
[2025-11-30 15:14:42] debug:    zh:zstack:unpi:writer: --> frame [254,8,33,51,1,1,0,97,0,0,0,12,119]
[2025-11-30 15:14:48] debug:    zh:zstack:znp: --> SREQ: SYS - nvRead - {"sysid":1,"itemid":1,"subid":97,"offset":0,"len":12}
[2025-11-30 15:14:48] debug:    zh:zstack:unpi:writer: --> frame [254,8,33,51,1,1,0,97,0,0,0,12,119]
[2025-11-30 15:14:54] debug:    zh:zstack:znp: --> SREQ: SYS - nvRead - {"sysid":1,"itemid":1,"subid":97,"offset":0,"len":12}
[2025-11-30 15:14:54] debug:    zh:zstack:unpi:writer: --> frame [254,8,33,51,1,1,0,97,0,0,0,12,119]
[2025-11-30 15:15:00] error:    z2m: Error while starting zigbee-herdsman
[2025-11-30 15:15:00] error:    z2m: Failed to start zigbee-herdsman
[2025-11-30 15:15:00] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-11-30 15:15:00] error:    z2m: Exiting...
[2025-11-30 15:15:00] error:    z2m: TypeError: Cannot read properties of undefined (reading 'payload')
    at AdapterNvMemory.readExtendedTableEntry (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:219:30)
    at AdapterNvMemory.readTable (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:365:28)
    at AdapterBackup.getAddressManagerTable (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-backup.ts:459:20)
    at AdapterBackup.createBackup (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-backup.ts:105:37)
    at ZStackAdapter.backup (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1032:16)
    at Controller.backup (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/controller/controller.ts:363:28)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/controller/controller.ts:185:13)
    at Zigbee.start (/app/lib/zigbee.ts:67:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
``` 

<details><summary>Full Z2M setup log</summary>
<p>

```
Starting Zigbee2MQTT without watchdog.
Onboarding page is available at http://0.0.0.0:8080/
[2025-11-30 15:14:37] info:     z2m: Logging to console, file (filename: log.log)
[2025-11-30 15:14:37] debug:    z2m: Can't load state from file /app/data/state.json (doesn't exist)
[2025-11-30 15:14:37] info:     z2m: Starting Zigbee2MQTT version 2.6.3 (commit #6e001b48ec278c6c141579c9eacc045f190fc289
)
[2025-11-30 15:14:37] info:     z2m: Starting zigbee-herdsman (6.3.2)
[2025-11-30 15:14:37] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":17703,\"extendedPanID\":[198,51,163,76,125,37,0,125],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":false,\"path\":\"tcp://10.10.10.34:6638\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":20}}"'
[2025-11-30 15:14:38] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":17703,"extendedPanID":[198,51,163,76,125,37,0,125],"channelList":[25]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"tcp://10.10.10.34:6638","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":20},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
[2025-11-30 15:14:38] info:     zh:zstack:znp: Opening TCP socket with 10.10.10.34:6638
[2025-11-30 15:14:38] info:     zh:zstack:znp: Socket connected
[2025-11-30 15:14:38] info:     zh:zstack:znp: Socket ready
[2025-11-30 15:14:38] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-11-30 15:14:38] debug:    zh:zstack:unpi:writer: --> buffer [239]
[2025-11-30 15:14:39] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-11-30 15:14:39] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-11-30 15:14:39] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-11-30 15:14:39] debug:    zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
[2025-11-30 15:14:39] debug:    zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
[2025-11-30 15:14:39] debug:    zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
[2025-11-30 15:14:39] debug:    zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
[2025-11-30 15:14:39] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-11-30 15:14:39] debug:    zh:zstack:znp: --> SREQ: SYS - version - {}
[2025-11-30 15:14:39] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,2,35]
[2025-11-30 15:14:39] debug:    zh:zstack:unpi:parser: <-- [254,10,97,2,2,1,2,7,1,209,254,52,1,0,116]
[... cut ...]
    at Zigbee.start (/app/lib/zigbee.ts:67:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
``` 

</p>
</details> 

To me this looks like Z2M is sending data to the UZG-01 and then at some point not receiving any more answer from the coordinator, making the zigbee-herdsman setup fail.

It seems like somehow my UZG-01 doesn't want to complete the setup. I think it's unlikely that the issue lies with ZHA or Zigbee2MQTT and rather with my UZG-01 unit or XZG.

xyzroe, 2026-11-15:
Closing: the reporter found the cause in very frequent HTTP health checks from the load balancer, and the other reporter has replaced the device. If the stalls still happen on [20261115](https://github.com/xyzroe/XZG/releases/tag/20261115), please open a new issue with logs.

closed 2026-11-15 by xyzroe