xyzroe/XZG #100: Unable to build XZG firmware - src/web.cpp:23:10: fatal error: webh/html/PAGE_VPN.html.gz.h: No such file or directory

victorhooi, 2024-07-29:
**Describe the bug**
Unable to build XZG

I've setup a Debian machine, and per the README have installed npm (from Debian repos) and platformio (using their [installer script](https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html)).

However, when I try to run `platformio run`, I get the following error:

``
$ platformio run
Processing prod-solo (platform: https://github.com/tasmota/platform-espressif32/releases/download/2023.07.00/platform-espressif32.zip; board: esp32-solo1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

██╗░░██╗███████╗░██████╗░
╚██╗██╔╝╚════██║██╔════╝░
░╚███╔╝░░░███╔═╝██║░░██╗░
░██╔██╗░██╔══╝░░██║░░╚██╗
██╔╝╚██╗███████╗╚██████╔╝
╚═╝░░╚═╝╚══════╝░╚═════╝░

Build: 20240730

Try to build WEB files

npm install --silent
sh: 1: npm: not found
*** Error 127
npx gulp xzg
sh: 1: npx: not found
*** Error 127

Finish building WEB files

██╗░░██╗███████╗░██████╗░
╚██╗██╔╝╚════██║██╔════╝░
░╚███╔╝░░░███╔═╝██║░░██╗░
░██╔██╗░██╔══╝░░██║░░╚██╗
██╔╝╚██╗███████╗╚██████╔╝
╚═╝░░╚═╝╚══════╝░╚═════╝░

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-solo1.html
PLATFORM: Espressif 32 (2023.7.0) > Espressif Generic ESP32-solo1 4M Flash
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduino-solo1 @ 2.0.11 
 - framework-arduinoespressif32 @ 3.0.3+sha.f5be003 
 - tool-esptoolpy @ 1.40602.0 (4.6.2) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.21.3
|-- CircularBuffer @ 1.4.0
|-- Ticker @ 4.4.0
|-- CronAlarms @ 0.1.0
|-- AsyncMqttClient @ 0.9.0
|-- AsyncTCP @ 1.1.1
|-- DallasTemperature @ 3.11.0
|-- HTTPClient @ 2.0.0
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- Preferences @ 2.0.0
|-- WiFi @ 2.0.0
|-- CCTools @ 0.0.6
|-- Ethernet @ 2.0.0
|-- OneWire @ 2.3.8
|-- WireGuard-ESP32 @ 0.1.5
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- WebServer @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio/build/prod-solo/src/config.cpp.o
Compiling .pio/build/prod-solo/src/const/hw.cpp.o
Compiling .pio/build/prod-solo/src/const/keys.cpp.o
Compiling .pio/build/prod-solo/src/etc.cpp.o
Compiling .pio/build/prod-solo/src/log.cpp.o
Compiling .pio/build/prod-solo/src/main.cpp.o
Compiling .pio/build/prod-solo/src/mqtt.cpp.o
Compiling .pio/build/prod-solo/src/per.cpp.o
Compiling .pio/build/prod-solo/src/web.cpp.o
src/web.cpp:23:10: fatal error: webh/html/PAGE_VPN.html.gz.h: No such file or directory
 #include "webh/html/PAGE_VPN.html.gz.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/prod-solo/src/web.cpp.o] Error 1
============================================================================================================ [FAILED] Took 6.98 seconds ============================================================================================================

Environment    Status    Duration
-------------  --------  ------------
prod-solo      FAILED    00:00:06.975
======================================================================================================= 1 failed, 0 succeeded in 00:00:06.975 =======================================================================================================
```

xyzroe, 2024-07-30:
npm install --silent
sh: 1: npm: not found
*** Error 127
npx gulp xzg
sh: 1: npx: not found
*** Error 127

Your machine doesn't have npm

xyzroe, 2026-11-15:
Closing: the build needs npm/Node.js. Current build instructions are in [AGENTS.md](https://github.com/xyzroe/XZG/blob/main/AGENTS.md).

closed 2026-11-15 by xyzroe