xyzroe/XZG #101: HTTP 502 on the General page of the Web interface when serving through reverse proxy

docstalek, 2024-08-01:
**Describe the bug**
When trying to access the General page, the page get stuck on loading (blue circular rotating)
I get an HTTP 502 on the /api?action=0&page=1

This only happens on the General page when serving the web interface through a Reverse proxy, Nginx Proxy Manager.

All other pages works without any issue.

When accessing directly through http://xzg.local/general, it works fine.

**Device information**
- Screenshot of Zigbee and Device tab from root page of device:
[image]

- JSON object returned on [http://xzg.local/api?action=1&param=all](http://xzg.local/api?action=1&param=all) 
`{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": true,
    "ethDhcp": true,
    "ethIp": "x.x.x.x",
    "ethMask": "255.255.255.0",
    "ethGate": "x.x.x.1",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "::",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-502F",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": true,
    "server": "x.x.x.x",
    "port": 1883,
    "user": "********",
    "pass": "********",
    "topic": "uzg",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 30
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": false,
    "webUser": "admin",
    "webPass": "",
    "fwEnabled": false,
    "fwIp": "x.x.x.x",
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 20,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 10,
    "hostname": "UZG-01",
    "timeZone": "Europe/Oslo",
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 1,
    "zbFw": "20240710",
    "updHour": "01:00",
    "updDays": "*",
    "autoIns": false
  }
}`
  
**To Reproduce**
Steps to reproduce the behavior:
1. Use Nginx Proxy Manager to serve the web interface through domain with SSL. e.g. https://zigbee.domain.com. --> Destination is http://xzg.local:80. Support for Websockets enabled.
2. Go to https://zigbee.domain.com/general
3. Loading icon appears.
4. Call to https://zigbee.domain.com/api?action=0&page=1 results in HTTP 500

**Expected behavior**
Expect the General page to show. Call to https://zigbee.domain.com/api?action=0&page=1 should result in HTTP 200.

DieMetRik, 2024-08-07:
I have the same issue 
All other works, except /general

SAOPP, 2024-09-04:
I use Traefik and I don’t see any similar problems, maybe the question concerns NPM?

Nitooo, 2025-02-14:
The happens becuase the header is to large for the default setting. You can go to the advanced tab and add this:
proxy_buffers 16 16k;
proxy_buffer_size 32k;
proxy_busy_buffers_size 64k;
proxy_max_temp_file_size 64k;

than it should just work fine.

xyzroe, 2026-11-15:
Closing: this is a reverse proxy buffer size issue, not a firmware bug. The nginx settings above (`proxy_buffers 16 16k; proxy_buffer_size 32k; proxy_busy_buffers_size 64k;`) fix it.

closed 2026-11-15 by xyzroe