xyzroe/XZG #104: Feature request: Option to disable polling for updates, and/or make it manual

asveikau, 2024-09-02:
**Is your feature request related to a problem? Please describe.**
Speaking personally I have a goal to minimize traffic to the external internet in my home devices.  I noticed that XZG has a periodic, default to 1am daily check for updates.  You can change the days and time but I didn't see a way to totally turn it off.

**Describe the solution you'd like**
An option to turn off checking for updates.
Perhaps also a button to manually check for updates, if you turn off the automatic check.

SAOPP, 2024-09-04:
Perhaps in the future the ability to disable automatic checking for updates of XZG will be added.

closed 2024-09-14 by xyzroe

hexajon-art, 2025-11-19:
Is it working as intended? I've disabled update checks, but I still get notified of updates:

[image]