xyzroe/XZG #105: UZG-01 unable to flash router firmware

habitats-tech, 2024-09-03:
I have several UZG-01 and I am unable to flash Zigbee firmware using the UI and Ethernet communication. Flashing the coordinator firmware is not an issue using ZigStarGW-MT. However, trying to flash router firmware seems to corrupt the device. Take the example below:

[image]

Trying to flash router firmware using the UI the update hangs on 66%.

[image]

[image]

Which leaves UZG-01 with corrupt firmware.

[image]

Using ZigStarGW-MT also corrupts the firmware.

[image]

[image]

The result is again corrupt firmware.

[image]

Does anyone have a solution to this problem.

habitats-tech, 2024-09-03:
I have attempted to erase NVRAM, but issue persists. Although it states cannot communicate over the serial port it can still be flashed using the ZigStarGW-MT flasher.

[image]

habitats-tech, 2024-09-04:
configHW.json

{
  "board": "UZG-01",
  "addr": 0,
  "pwrPin": 5,
  "mdcPin": 23,
  "mdiPin": 18,
  "phyType": 0,
  "clkMode": 3,
  "btnPin": 35,
  "btnPlr": 1,
  "uartSelPin": 33,
  "uartSelPlr": 1,
  "ledModePin": 12,
  "ledModePlr": 1,
  "ledPwrPin": 14,
  "ledPwrPlr": 1,
  "zbTxPin": 4,
  "zbRxPin": 36,
  "zbRstPin": 16,
  "zbBslPin": 32
}

SAOPP, 2024-09-04:
Try the following:

1. reboot the coordinator
2. disconnect all connections (socket and mqtt)
3. retry flashing

habitats-tech, 2024-09-04:
All my actions were performed on a device which had no connections to anything except Ethernet.

cloudbr34k84, 2024-09-08:
Im trying to flash an older version of Z2M, uisng the new interface, and it says its flashed and reboots, but it still says 20240707 when i chose 20240316
[image]

[image]

SAOPP, 2024-09-08:
You are talking about usp firmware.

SAOPP, 2024-09-08:
Zigbee firmware is located on the status page under the heading called Zigbee.

SAOPP, 2024-09-08:
> an older version of Z2M

Btw, wtf this mean:question:

cloudbr34k84, 2024-09-08:
just realised this is a router issue post. ill delete and add a new issues

SAOPP, 2024-09-08:
> just realised this is a router issue post. ill delete and add a new issues

Just share ya full screenshot of status page.

SAOPP, 2024-09-08:
Does the firmware process complete correctly?

cloudbr34k84, 2024-09-08:
> > just realised this is a router issue post. ill delete and add a new issues
> 
> Just share ya full screenshot of status page.

[image]

SAOPP, 2024-09-08:
So u tried to flash 20230416 - it's correct?

cloudbr34k84, 2024-09-08:
> Does the firmware process complete correctly?

yep

> So u tried to flash 20240316 - it's correct?

yep

SAOPP, 2024-09-08:
Have u tried opening the coordinator's web interface in incognito mode or from another browser?

Try a couple of simple steps, clear the cache of the current page (you never know), try again, reboot the coordinator and immediately start the firmware process again.

cloudbr34k84, 2024-09-08:
> Have u tried opening the coordinator's web interface in incognito mode or from another browser?
> 
> Try a couple of simple steps, clear the cache of the current page (you never know), try again, reboot the coordinator and immediately start the firmware process again.

video of the whole process, unless im doing something wrong. have cleared cache, will try your steps
https://github.com/user-attachments/assets/e8e4f37d-430d-41e3-8b77-5d7153492314

cloudbr34k84, 2024-09-08:
Ok opened web UI in incognito
cleared cache
restarted device
immediately tried OTA firmware update: to 20240316
Im now getting "No response; check if IP or USB mode changed."
Serial settings: says Network Mode

SAOPP, 2024-09-08:
> video of the whole process, unless im doing something wrong. have cleared cache, will try your steps https://github.com/user-attachments/assets/e8e4f37d-430d-41e3-8b77-5d7153492314

Did the process complete automatically or did you click the page refresh button?

cloudbr34k84, 2024-09-08:
> > video of the whole process, unless im doing something wrong. have cleared cache, will try your steps https://github.com/user-attachments/assets/e8e4f37d-430d-41e3-8b77-5d7153492314
> 
> Did the process complete automatically or did you click the page refresh button?

i let it complete

SAOPP, 2024-09-08:
Looks like HG devices overdeep, @mercenaruss @xyzroe any comments?

xyzroe, 2024-09-08:
I have no ideas. Maybe someone else could check the same situation?

SAOPP, 2024-09-08:
> i let it complete

As one of the options, I can suggest using the web installer, and flashing the device again, and then try flashing the zigbee chip again...

cloudbr34k84, 2024-09-08:
> > i let it complete
> 
> As one of the options, I can suggest using the web installer, and flashing the device again, and then try flashing the zigbee chip again...

one step ahead of yeah

cloudbr34k84, 2024-09-08:
well i was finally able to to use the multi tool to install the previous FW. 
[image]

habitats-tech, 2024-09-08:
The only reliable way to flash coordinator firmware is using the ZigStarGW-MT tool.

Can anyone point me to the most recent router firmware release for UZG-01 to try flash it using the ZigStarGW-MT tool to UZG-01.

Using the webUI is a waste of time as it does not work.

cloudbr34k84, 2024-09-08:
> The only reliable way to flash coordinator firmware is using the ZigStarGW-MT tool.
> 
> Can anyone point me to the most recent router firmware release for UZG-01 to try flash it using the ZigStarGW-MT tool.
> 
> Using the webUI is a waste of time as it does not work.

Then why is it there..

habitats-tech, 2024-09-08:
I have no idea why it works for some people. For me it has never worked, unlike SLZB which works flawlessly every time.

I am not here to debate why not. I am here to resolve an issue.

Can ANYONE point me to the latest router firmware for UZG-01 (it has to be router firmware not coordinator firmware).

habitats-tech, 2024-09-08:
For anyone reading this. Whenever you flash Zigbee firmware, you should always reboot the device (not just Zigbee) to get the most current status of the device. It works every time.

habitats-tech, 2024-09-11:
So I think I know what the issues updating FW through the webUI are.

My primary network sits behind upstream and downstream proxies and other security features. When I try to update the UZG-01 FW on this network it always fails while SLZB devices have no such issues.

Setting up an open network so far have been able to update UZG-01 FW successfully through the webUI.

cloudbr34k84, 2024-09-11:
So I was successful on updating the firmware as well via the UI, but I was
on apple MacBook running chrome..
I don't have any proxies so who knows

dmz006, 2025-02-27:
I have a UZG-POE 0.5 that I decided to upgrade yesterday; now it's running 20240914 for ESP but when I try to install zigbee I get:
[2405185] | Start Zigbee flashing
[2405189] | ZB flash https://raw.githubusercontent.com/xyzroe/XZG/zb_fws/ti/coordinator/CC1352P7_coordinator_20230507.bin @ 115200
[2408254] | Erase completed!
[2410148] | Error initializing flash process
[2411151] | Erase completed!
[2411302] | Failed to download file, HTTP code: -7

... - repeats a few times

[2438682] | Failed to flash Zigbee
[2438689] <- fe 00 06 00 f1 e7 e5 fe

I have tried multiple images and all are showing the same.

Now since zigbee image is erased it doesn't have any firmware for zigbee and it's not working.

I've tried attaching to local USB, lots of different things; but can't seem to get zigbee firmware loaded.

based on UZG-01 web site I should use 'UZG-01 v0.2+ - Use firmware with a tag "CC1352P7"' - so tried CC1352P7_coordinator_20240710.hex using ZigStar MT:
Opening port socket://192.168.1.19:6638, baud 500000
Reading data from /home/dmz/Downloads/CC1352P7_coordinator_20240710.hex
Firmware file: Intel Hex
Connecting to target...
Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'

with debug log on xzg :
[402516] | ZB enable BSL
[403565] | Now you can flash CC2652!
[403573] <- 06 a8
[408622] | Connected client 2 from 192.168.1.51
[408969] -> 55 55

Is there any way to adjust the speed? zigbee is set to 115200. When trying to flash zigbee in XZG I notice a b=115200 ; but multi tool doesn't seem to have an option.

55 55 on other searches said it was a zigbe chip issue.

Not sure if thats part of the problem; or what i'm doing wrong trying to reflash zigbee  Any suggestions welcome.

habitats-tech, 2025-02-27:
The best way to upgrade when UZG refuses to update the image is:

1. Download the attached zigbee FW image (or download the 20240710 version)

[CC1352P7_coordinator_99240915.zip](https://github.com/user-attachments/files/19009044/CC1352P7_coordinator_99240915.zip)

2. Use the ZigStarGW-MT tool to update the image. You can update the image over the network. Type the IP address of the UZG followed by port 6638 -> select file (the hex downloaded earlier) -> enable ERASE, WRITE, VERIFY -> Start

[image]

habitats-tech, 2025-02-27:
Before you select the firmware, can you try to read the IEEE.

When timeout error this means no communication with the device. The way I solved it in the past was to connect a PC directly to the UZG through a simple switch, bypassing the rest of the network.

I have found the hard way the UZG will not play ball on some networks, but never found the time to find out why.

habitats-tech, 2025-02-27:
It also helps if the computer you are using to connect to the network is connected through wired Ethernet (not WiFi).

dmz006, 2025-02-27:
> Before you select the firmware, can you try to read the IEEE.
>
> When timeout error this means no communication with the device. The way I solved it in the past was to connect a PC directly to the UZG [...]

I get the same "Timeout waiting for ACK/NACK after 'Sync (0x55 0x55)'" when trying to read the IEEE.

I may need to find a windows machine and try the TI firmware tool.

HASBULATUDALOY, 2025-03-15:
Hello everyone. dmz006, I have the same problem as you. Did you manage to solve it?

guillaume-bzh, 2025-09-13:
Hello everyone,
I had exactly the samedi problem on my first uzg after an upgrade, 18 month later: same thing on my second uzg (upgrade because the current version has zigbee2mqtt frozen after some hours without possibilities to script it).
I saw similar case with zigate, and capabilities to force bsl on boot with 10uF capacitor between 2 pin of ESP, but I don't know if you can do that on uzg with esp wroom s2.
Thanks for your help (I need to reapaired all my zigbee devices.....).
Guillaume.

guillaume-bzh, 2025-09-13:
Oh my god, there are Alive ! 
I find a tips where you put a 10uf capacitor between pin "gnd" (pin 1) and "en"  (pin 3) and using Web install flasher online. 
First I select "see logs" and after install Fw ans it works ! 
I have blocked the reste button to be sure but I don't know if it works. 
[image]
I wish it can help you !

xyzroe, 2025-09-14:
@guillaume-bzh Hi. 
I'm glad you were able to fix your problem. 
I want to be clear: this issue is related to being unable to flash router firmware (which is inside the CC2652P*). 
According to your message, you had a problem flashing XZG firmware (which is inside the ESP32). 
Could you please explain this more clearly?

xyzroe, 2025-09-14:
But dmz006 had problem with CC2652, not ESP32...

guillaume-bzh, 2025-09-14:
Hum... I had the same error message : Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
But impossible for me to connect to the Uzg-01 without this method...
It permit to me to unlock and recover my Uzg-01 (including zigbee Fw)

closed 2025-11-21 by xyzroe

xyzroe, 2026-02-05:
What was the error when using XZG Multi-Tool?

xyzroe, 2026-02-06:
May I ask for some details? Logs or screen shots?

1NoOneKnowsMe1, 2026-02-06:
Apologies, I don’t have additional logs available at the moment.
The XZG Multi-Tool did not expose useful logging in my setup.
I do have a second identical device and plan to repeat the test once another coordinator arrives, and I will share the findings.