xyzroe/XZG #106: Feature request: the "old" firmware give the temperature of the device

alainvdu69, 2024-09-03:
I've just flashed my ZigStarLanGW and i've noticed that there is no more the temperature of the device

SAOPP, 2024-09-04:
What sensor are we talking 'bout in this case? The coordinator temperature sensor transmitted via the mqtt?

alainvdu69, 2024-09-04:
yes

SAOPP, 2024-09-04:
But I don’t see such a sensor in ZigStar, can u be more specific what you’re talking about?

SAOPP, 2024-09-04:
Here is all what I've in from my ZigStar GW POE:

[image]

What the sensor u're talking?

alainvdu69, 2024-09-04:
in home assistant i have this :

sensor.cpu_temperature
ZigStarGW ESP Temperature
25.56

alainvdu69, 2024-09-04:
well ... it's strange ... this sensor come back with the XZG

alainvdu69, 2024-09-04:
so, i close ...

alainvdu69, 2024-09-04:
didn't understand what happened ... the sensor come back on the device page in home assistant 1 day after the upgrade to XZG firmware ...

closed 2024-09-04 by alainvdu69