xyzroe/XZG #108: Frequent device reboots. HamGeek HMG-01 Plus POE

codinv, 2024-09-12:
**Describe the bug**
HamGeek HMG-01 Plus POE reboots every day

**Device information**
[image]

{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": true,
    "ethDhcp": true,
    "ethIp": "::",
    "ethMask": "255.255.255.0",
    "ethGate": "::",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "::",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-15B3",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": true,
    "server": "192.168.xx.xx",
    "port": 1883,
    "user": "xxxxxx",
    "pass": "xxxxxx",
    "topic": "HMG-01Plus",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 15
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": false,
    "webUser": "",
    "webPass": "",
    "fwEnabled": false,
    "fwIp": "0.0.0.0",
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 13,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 1,
    "hostname": "HMG-01Plus",
    "timeZone": "Europe/Moscow",
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 1,
    "zbFw": "20240710",
    "updHour": "02:00",
    "updDays": "7",
    "autoIns": false
  }
}
  
**To Reproduce**
turn on and wait, the device will reboot itself within 24 hours

**Additional context**
==== and much earlier ===

[2024-09-08 20:14:30] info: 	zh:zstack:znp: Socket error
[2024-09-08 20:14:30] info: 	zh:zstack:znp: Port closed
[2024-09-08 20:14:30] info: 	zh:zstack:znp: closing
[2024-09-08 20:14:30] error: 	z2m: Adapter disconnected, stopping
[2024-09-08 20:14:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'

[2024-09-09 20:29:45] info: 	zh:zstack:znp: Socket error
[2024-09-09 20:29:45] info: 	zh:zstack:znp: Port closed
[2024-09-09 20:29:45] info: 	zh:zstack:znp: closing
[2024-09-09 20:29:45] error: 	z2m: Adapter disconnected, stopping
[2024-09-09 20:29:45] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'

[2024-09-10 20:29:41] info: 	zh:zstack:znp: Socket error
[2024-09-10 20:29:41] info: 	zh:zstack:znp: Port closed
[2024-09-10 20:29:41] info: 	zh:zstack:znp: closing
[2024-09-10 20:29:41] error: 	z2m: Adapter disconnected, stopping
[2024-09-10 20:29:41] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'

[2024-09-11 21:04:08] info: 	zh:zstack:znp: Socket error
[2024-09-11 21:04:08] info: 	zh:zstack:znp: Port closed
[2024-09-11 21:04:08] info: 	zh:zstack:znp: closing
[2024-09-11 21:04:08] error: 	z2m: Adapter disconnected, stopping
[2024-09-11 21:04:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'

[2024-09-12 21:02:22] info: 	zh:zstack:znp: Socket error
[2024-09-12 21:02:22] info: 	zh:zstack:znp: Port closed
[2024-09-12 21:02:22] info: 	zh:zstack:znp: closing
[2024-09-12 21:02:22] error: 	z2m: Adapter disconnected, stopping
[2024-09-12 21:02:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'

xyzroe, 2024-09-14:
Try to disable cron based update availability check

codinv, 2024-09-14:
Okay, I just updated the firmware and set "Disabled" in the update check interval
But, I would like to note that earlier I had already configured an update only on Sunday, and at two o'clock in the morning, but the device reboots daily and a few hours earlier.
Anyway, thank you for the update, I'll let you know if it will help with my problem

codinv, 2024-09-14:
@xyzroe does not help :(
2024-09-14 18:28:10 - update firmware to 20240914 and reboot device
2024-09-14 18:29:49 - set disable check update and reboot device
2024-09-14 21:01:28 - independent reboot of the device

[image]
[image]

codinv, 2024-09-15:
and again 

[2024-09-15 14:39:07] info: 	zh:zstack:znp: Socket error
[2024-09-15 14:39:07] info: 	zh:zstack:znp: Port closed
[2024-09-15 14:39:07] info: 	zh:zstack:znp: closing
[2024-09-15 14:39:07] error: 	z2m: Adapter disconnected, stopping
[2024-09-15 14:39:07] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'

untergeek, 2024-09-19:
I have the exact same problem on my UZG-01, spontaneously rebooting with no real explanation why. I see that the uptime matches roughly from when the zigbee2mqtt logs report the `Socket error`.

untergeek, 2024-09-19:
Oh, and cron checks are disabled.

xyzroe, 2024-09-20:
> Something is DEFINITELY weird here because they both rebooted at effectively the same amount of elapsed time:

What are you trying to say? What is wrong? You click reboot - you have reboot. 

> So, what gives? What waits somewhere between 6 and 8 hours and then just insta-kills the TCP connections to the device, causing zigbee2mqtt to display, `Socket error` and restart?

When the device is rebooting the TCP server on XZG kills itself. It's ok. 

But we need to figure out why your device is rebooting. Are you using DCHP on your device? Could you try to switch to static IP?

untergeek, 2024-09-20:
What I was saying is that the _non-forced_ reboots happened at the same time after each box was manually rebooted. I thought that to be rather odd.

I will try to set the IPs manually. And I will disable WiFi and go wired-only, just to isolate.

codinv, 2024-09-20:
> But we need to figure out why your device is rebooting. Are you using DCHP on your device? Could you try to switch to static IP?
I have a static IP by MAC address for HMG-01 on my router, but the device still reboots, sometimes even more than once a day.
A few days ago I moved all zigbee devices to another router, but that didn't help either, there were several reboots again.
It seems to me that reboots occur when the device tries to access some resource on the external network, but cannot do so. If you completely block the device from accessing the Internet, reboots still occur.
Is it possible to somehow organize logging on the device to find out the exact reason for the reboot?

xyzroe, 2024-09-20:
> I have a static IP by MAC address for HMG-01 on my router, but the device still reboots, sometimes even more than once a day. 
No, I'm talking about real static IP on device. Not on DHCP server

xyzroe, 2024-09-20:
> Is it possible to somehow organize logging on the device to find out the exact reason for the reboot?

Of course, project is open source, anyone can make a such option.

xyzroe, 2024-09-20:
> disable WiFi and go wired-only

It's the first thing that you must to do in case of any problems.

xyzroe, 2024-09-20:
I know of a case where a special DHCP server configuration causes the device to reboot every time a DHCP lease is updated. Compare reboot frequency and lifetime of DHCP entry in your router

untergeek, 2024-09-20:
> > disable WiFi and go wired-only
> 
> It's the first thing that you must to do in case of any problems.

I was using both wired and wireless. Each was on a different subnet. Now wired has a statically assigned IP address (not via DHCP) and WiFi is off. Will report back findings when I see the next reboot (or if I don't see one).

untergeek, 2024-09-20:
> I know of a case where a special DHCP server configuration causes the device to reboot every time a DHCP lease is updated. Compare reboot frequency and lifetime of DHCP entry in your router

Interesting. My DHCP server's `default-lease-time` is `7200` seconds, or 2 hours, with the `max-lease-time` set to `86400` seconds (1 day). The reboots were not happening every 2 hours, but more frequently than once per day. Is that special DHCP server configuration 100% guaranteed to trip each time. 🤷‍♂️ 

I guess we'll wait and see in my case.

codinv, 2024-09-20:
> then when you turn off full access to the Internet, the device will not be able to turn on, because during startup, the presence of updates is checked.

when the internet is full turned off for only this device, it boots normally. the only thing I changed was to assign my own NTP server so that the time would be set correctly on the device

> Compare reboot frequency and lifetime of DHCP entry in your router

lease 25200 seconds (every 7 hours)
and since reboots are mostly once every 24 hours, it turns out that this is not the reason

> Of course, project is open source, anyone can make a such option.

hmm.. this looks very difficult =)

untergeek, 2024-09-24:
So, status report. Both of my zigbee2mqtt instances spontaneously rebooted at least one time. However, this is what I see today:

```
NAME                READY   STATUS    RESTARTS      AGE
pod/zigbee2mqtt-0   1/1     Running   20 (4d ago)   5d19h
```

and

```
NAME                READY   STATUS    RESTARTS      AGE
pod/zigbee2mqtt-0   1/1     Running   10 (4d ago)   5d19h
```

Static wired IP and no WiFi seem to be more stable. I did not add or remove any devices in that timeframe, or any other activity besides just "running." I will be adding another round of devices soon, however, so we will see if that activity changes anything.

I will continue to monitor.

rsdes, 2024-10-02:
I experience the same issue, Reboots after 4h till 24h uptime.

Gerätemodell: HG POE Plus
FW-Revision: 20241001
HW-Version: ESP32-D0WD-V3

Static IP, no WiFi, no Update Check, local NTP Server, local DNS Server, not one Zigbee device connected just idling.

The Console shows the following output, maybe someone can read something from this output. 

`
E (43887268) wifi_init_default: netstack cb reg failed with 12308
E (43891258) wifi_init_default: netstack cb reg failed with 12308
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (Tmr Svc) 
Core  0 register dump:
PC      : 0x40091fa3  PS      : 0x00060b36  A0      : 0x8008f72d  A1      : 0x3ffcf570  
A2      : 0x3ffd580c  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060b23  
A6      : 0x00060b20  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x0000003f  
A14     : 0x007d580c  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008b4a5  LEND    : 0x4008b4b5  LCOUNT  : 0xfffffffe  

Backtrace: 0x40091fa0:0x3ffcf570 0x4008f72a:0x3ffcf5b0 0x4008f7ec:0x3ffcf5f0 0x4008552d:0x3ffcf610 0x40085639:0x3ffcf640 0x4010d0da:0x3ffcf660 0x4010c357:0x3ffcf680 0x4010b905:0x3ffcf6a0 0x4008b085:0x3ffcf6c0 0x40089926:0x3ffcf6e0 0x400899b0:0x3ffcf700 0x40089e9d:0x3ffcf730 0x401a7697:0x3ffcf760 0x401a2b73:0x3ffcf780 0x401a2cf1:0x3ffcfa90 0x401b4fc5:0x3ffcfac0 0x4013a6da:0x3ffcfaf0 0x4013a777:0x3ffcfb20 0x4011ad1d:0x3ffcfba0 0x4011ad7e:0x3ffcfbc0 0x400f0504:0x3ffcfbe0 0x400d9f3a:0x3ffcfc00 0x400dda9e:0x3ffcfc30 0x400e0f0b:0x3ffcfc50 0x40091b43:0x3ffcfce0

ELF file SHA256: 2e8b942b1a999a93

Rebooting...
ets Jul 29 2019 12:21:46`

Can I increase the Debug Level to have more output?

Thanks for your efforts.

untergeek, 2024-10-02:
> wifi_init_default: netstack cb reg failed with 12308

That seems odd, if WiFi is disabled. What's being initialized?

rsdes, 2024-10-02:
> > wifi_init_default: netstack cb reg failed with 12308
> 
> That seems odd, if WiFi is disabled. What's being initialized?

I agree, but thats what I see in the logs.

Never touched the WIFi settings.
[image]

xyzroe, 2024-10-02:
Thanks for feedback. Already found this bug, trying to fix.

turbolooser, 2024-10-07:
Would this bug also affect my zigbee-gw-lan HW?
I do see super frequent restarts also... :-(
Any hint other than statis ip , disabling update check etc?

l3zha, 2025-01-03:
did you manage to find a solution? I tried to downgrade the firmware, the reboots disappeared, but the entire network began to fail

l3zha, 2025-02-05:
> I'm using the original UZG-01 https://uzg.zig-star.com/product/ and don't have such problems

После вашего поста, купил оригинальное устройство вместо hamgeek, но ребуты продолжились

After your post, I bought the original device instead of hamgeek, but the reboots continued

l3zha, 2025-02-12:
I turned off mqtt and the reboots stopped

closed 2025-02-12 by xyzroe

xagos, 2025-02-17:
I have exactly the same problem with my hamgeek board - I am going to disable mqtt and see if the reboot stops.
Thanks for your work and your support...

hubertbanas, 2025-02-17:
@xagos, do you have a timeline when can we expect new release?

xyzroe, 2025-02-17:
> [@xyzroe](https://github.com/xyzroe), do you have a timeline when we can expect new release?

No, at least not now.

0BanZai0, 2025-03-31:
I'm on a HMG-01 POE as well. I'm not having reboots but sometimes Zigbee2Mqtt loses the connection to the coordinator for a brief moment causing it to send all the last known states of my devices to Home Assistant after re-establishing the connection. This causes a lot of trouble with switches that repeat their keypresses... Could this be related?

Imaginous, 2025-05-22:
I have an UZG-01 (CC2652P7) with firmware 20241001 (the firmware it shipped with).
It also randomly reboots between 11 hours  and a day or 2.

- Checking for updates is disabled
- MQTT connected
- Using zigbee2mqtt with zstack firmware 20240710

Just for your information.

xyzroe, 2025-05-23:
Disable MQTT

Imaginous, 2025-05-23:
Do you know what in the MQTT connection might cause the reboots?

xyzroe, 2025-05-23:
memory overflow

0BanZai0, 2025-05-26:
I'm not having MQTT enabled but my device restarted some minutes ago which caused devices in home assistant to do strange things like opening my blinds.
Where can I find a debug log to see what might have caused this reboot?

fliespl, 2025-12-29:
@xyzroe Thanks for getting back so quickly. Totally understandable - open source can be demanding, and your work is still genuinely appreciated. Wishing you good energy for whatever comes next, and thanks again!