xyzroe/XZG #111: After update to last Beta: File not found

Diddlik, 2024-10-09:
**Describe the bug**
After updating to the latest beta when calling up Web-UI, the error message appears: File not found

**Device information**
- Olizig
- {
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "0.0.0.0",
    "wifiMask": "255.255.255.0",
    "wifiGate": "0.0.0.0",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": true,
    "ethDhcp": false,
    "ethIp": "192.168.1.40",
    "ethMask": "255.255.255.0",
    "ethGate": "192.168.1.1",
    "ethDns1": "192.168.1.50",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "0.0.0.0",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-EB10",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": true,
    "server": "192.168.1.50",
    "port": 1883,
    "user": "*****",
    "pass": "******",
    "topic": "Olizig",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 30
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": false,
    "webUser": "",
    "webPass": "",
    "fwEnabled": false,
    "fwIp": "0.0.0.0",
    "fwMask": "0.0.0.0",
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 20,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 1,
    "hostname": "Olizig",
    "timeZone": "Europe/Berlin",
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 1,
    "zbFw": "20240710",
    "updHour": "01:00",
    "updDays": "*",
    "autoIns": false
  }
}

Eucliwood090, 2024-10-10:
I have the same problem and can't go back to another version...

Eucliwood090, 2024-10-11:
no one ?

cnstudio, 2024-10-12:
I have the same problem too! 
happen when upgrade from 20240914 to 20241001, the web UI only show "file not found",
and device keeping reboot itself again and again.

so I can NOT use ZigStarGW-MT tool to flash firmware to other version,
because when ZigStarGW-MT upload firmware file to about 10% ~ 15%, then the device reboot itself.

mihsu81, 2024-10-12:
You can flash it over serial (USB) using the [Web Flasher](https://xzg.xyzroe.cc/install/)
Worked fine for me.
P.S. Keep in mind that you'll lose your previous settings after flashing but you can copy them from the serial logs of the Web Flasher before flashing.

Eucliwood090, 2024-10-12:
How did you do it because it's impossible, the flash doesn't start, did you touch the switch? Do a special manipulation?

Diddlik, 2024-10-12:
Simply connect to the PC via USB and start the flash process

Eucliwood090, 2024-10-12:
i have https://zig-star.com/projects/zigbee-gw-lan/

Eucliwood090, 2024-10-12:
E (17545) wifi_init_default: netstack cb reg failed with 12308
13740 [printLogMsg] [RCP] FW: Unknown! Check serial speed!
                                                          13740 [setup] [RCP] FW                                                                                                                                  : 0
   13740 [setup] [ESP] FW: 20241001
                                   13751 [setupCoordinatorMode] Network
                                                                       13860 [mD                                                                                                                                  NS_start] mDNS responder started on ZigStarGW.local
                                                   13895 [setup]
                                                                 {
  "system-vars": {
    "hwBtnIs": false,
    "hwLedUsbIs": false,
    "hwLedPwrIs": false,
    "hwZigbeeIs": true,
    "connectedClients": 0,
    "socketTime": 0,
    "connectedEther": false,
    "apStarted": false,
    "wifiWebSetupInProgress": false,
    "vpnWgInit": false,
    "vpnWgConnect": false,
    "vpnWgPeerIp": "0.0.0.0",
    "vpnWgCheckTimeKey": 0,
    "vpnHnInit": false,
    "mqttConn": false,
    "mqttReconnectTime": 0,
    "mqttHeartbeatTime": 0,
    "disableLeds": false,
    "zbFlashing": false,
    "deviceId": "XZG-5291",
    "espUpdAvail": false,
    "rcpUpdAvail": false
  }
}
 13946 [setup] done
                   E (20549) wifi_init_default: netstack cb reg failed with 1230                                                                                                                                  8
E (24546) wifi_init_default: netstack cb reg failed with 12308
20374 [mDNS_start] mDNS responder started on ZigStarGW.local

Diddlik, 2024-10-12:
than you have to use ttl converter to flash esp32

pullasuti, 2024-12-10:
Any ideas what tools would allow me to flash the device? I have a bootlooping GW as well. Can't use ZigstarGW-MT, since it and cc2538-bsl just keep giving the same error (Target returned: 0x43, Invalid address. This happens over ethernet and usb). Esptool.py won't detect the device either , using usb or a ttl adapter. And the web flasher is a no go, too.

pullasuti, 2024-12-10:
Ok, I was finally able to flash the chip with instructions from [here](https://community.home-assistant.io/t/how-i-installed-esphome-on-the-wt32-eth01/359027)

xyzroe, 2024-12-13:
https://github.com/xyzroe/XZG/issues/131#issuecomment-2540876772

xyzroe, 2026-11-15:
Closing, same cause as #136: the file system (web files) wasn't updated together with the firmware. See https://github.com/xyzroe/XZG/issues/131#issuecomment-2540876772.

closed 2026-11-15 by xyzroe