xyzroe/XZG #115: UZG-01 network break down

TangoAlfaNeroYpsilon, 2024-10-26:
Hallo,
Trying to install UZG-01 and ran into release update, but re-directed to XZG. Looks like the device is alive (Static IP, MQTT, and Zigbee flashing ON/OFF (1 device), but does not recognizes TAB -> Coordinator, with msg : "Changing the firmware on a running network cause the network break down!". No device was coupled yet.
When clicking the red square "Coordinator" + "I'm sure"... then it contineous with Zigbee OTA update + will disconnect ALL socket + "I'm sure"..........>   "going to flash CC1352P7_coordinator_20240710.bin"... and after some minutes ends with "ZigBee OTA update : Failed".
[UZG prints.docx](https://github.com/user-attachments/files/17531112/UZG.prints.docx)
Prints attached.
Help appreciated.
Cheers,
T

xyzroe, 2024-10-26:
Try to disable MQTT on the device, reboot the device, try again to flash zigbee firmware

TangoAlfaNeroYpsilon, 2024-10-26:
MQTT disabled. With same procedure : same OTA update Failed.
I have downloaded "UZG-01.bin". 
Your suggestion "flash zigbee firmware"... is that : Tools-> Firmware update -> choose File + update ?
T

TangoAlfaNeroYpsilon, 2024-10-27:
Hallo xyzroe,
to keep you posted : UZG-01.bin loaded.
MQTT connected, Zigbee Socket client connected, 1 IKEA device paired with Zigbee2MQTT-frontend.
:)
1 concern : IP/zigbee shows as Role still "Changing the firmware on a running network cause the network break down!" and the red button "Coordinator" is not highlighted. 
Looks like everything is working well... except possibly for running updates.

Cheers,
T

xyzroe, 2024-10-27:
I can't get your point. There are no files named "UZG-01.bin" in this repository.

TangoAlfaNeroYpsilon, 2024-10-28:
My point is that for zigbee i am "newbee", and setting up Node-Red/ZigBee2MQTT/SONOFF Bridge... failed, though the TASMOTA flashing worked well.
Then I went for UZG-01, and followed the web site instructions but was redirected to XZG. And there it failed again as i described in the first post.
So I tried again via site of UZG-01 and did the OTA Update... that guided me to download UZG-01.bin... which update was immed a succesfull.
Node-Red, AEDES, FrontEnd of ZigBee2MQTT and browser on the UZG-01 via XZG-webpage... all work fine.
ZigBee is on release 20240710 and ESP on 20240519.

So I guess not to bother about the Role-warning.
I also noticed there are many logs. Every 5 secs a new folder with log.log. Is there a way to limit this ,  
 
Cheers,
T

xyzroe, 2026-11-15:
Closing: the reporter got the device working.

closed 2026-11-15 by xyzroe