xyzroe/XZG #122: sensor.uzg_01_esp_temperature no longer has a state class

spitfire, 2024-11-05:
**Describe the bug**
Home Assistant states that sensor.uzg_01_esp_temperature (the sensor provided from my ZigBee gateway running XZG) no longer has a state class. This prevents it from being able to have long term statistics stored. 

**To Reproduce**
Connect your GW running XZG to Home Assistant using MQTT, look at the fixes section in settings

**Expected behavior**
Sensors provided by XZG have a state class

xyzroe, 2024-11-05:
version?

spitfire, 2024-11-05:
20240914

xyzroe, 2024-11-06:
You can try the latest

spitfire, 2024-11-06:
> You can try the latest 

I was not brave enough to test it at first (warnings on the release page!), but it does seem to fix this issue (statistics seem to be recorded and HA does not complain anymore).

xyzroe, 2026-11-15:
Closing: confirmed fixed since 20241001 by the reporter.

closed 2026-11-15 by xyzroe

spitfire, 2026-11-15:
> Closing: confirmed fixed since 20241001 by the reporter.

Can you make it an official release then?

xyzroe, 2026-11-16:
there's some pre-releases with that fix. look at  the releases page.