xyzroe/XZG #123: Zigbee2mqtt and XZG-01

allseven, 2024-11-05:
My zigbee2mqtt installation on HAOS with a XZG-01 coordinator has suddenly stopped working with zigbee2mqtt refusing to start.

I've gone back to a working backup of HA but that's not fixed it.

Hours of fiddling around has probably killed it off big time but do any of these screenshots offer any clues as to what the issue might be please?

As the HA backup restore hasn't fixed the problem, could my XZG-01 configuration the issue?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

xyzroe, 2024-11-05:
on your screenshot USB mode is active, not LAN

allseven, 2024-11-05:
Doh!! Thank you.I don't know how that got changed. 🤔🙄Best wishesFrank

closed 2024-11-12 by xyzroe