xyzroe/XZG #124: Failed to build filesystem. Exiting.

lyt580231, 2024-11-07:
Processing prod-dual (platform: espressif32 @ 6.8.1; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option

██╗░░██╗███████╗░██████╗░
╚██╗██╔╝╚════██║██╔════╝░
░╚███╔╝░░░███╔═╝██║░░██╗░
░██╔██╗░██╔══╝░░██║░░╚██╗
██╔╝╚██╗███████╗╚██████╔╝
╚═╝░░╚═╝╚══════╝░╚═════╝░

Build: 20241107

Try to build WEB files

Git commit SHA saved to ./data/x/commit: ce1eb6b
npm install already executed. Skipping.
npx gulp xzg
[21:41:50] Using gulpfile ~\Desktop\xzg\XZG\tools\webfilesbuilder\gulpfile.js
[21:41:50] Starting 'xzg'...
[21:41:50] Starting 'stylesConcat'...
[21:41:50] Starting 'scriptsgz'...
[21:41:50] Starting 'imggz'...
[21:41:50] Starting 'htmlgz'...
[21:41:50] Starting 'jsongz'...
[21:41:53] Finished 'imggz' after 3.86 s
[21:41:55] Finished 'stylesConcat' after 5.29 s
[21:41:55] Finished 'scriptsgz' after 5.77 s
[21:41:55] Finished 'htmlgz' after 5.86 s
[21:41:55] Finished 'jsongz' after 5.86 s
[21:41:55] Finished 'xzg' after 5.87 s

Finish building WEB files
Using existing filesystem size from C:\Users\Dream.LEE\Desktop\xzg\XZG\src\websrc\size.fs: 364544 bytes
Path to mklittlefs tool is not set or invalid: C:\Users\Dream.LEE\.platformio\packages\tool-mklittlefs\mklittlefs
Failed to build filesystem. Exiting.

xyzroe, 2024-11-07:
> Path to mklittlefs tool is not set or invalid: C:\Users\Dream.LEE.platformio\packages\tool-mklittlefs\mklittlefs
I think this is the problem.

lyt580231, 2024-11-08:
> > Path to mklittlefs tool is not set or invalid: C:\Users\Dream.LEE.platformio\packages\tool-mklittlefs\mklittlefs
> 
> I think this is the problem.

I don't know how to set mklittlefs tool ，C:\Users\Dream.LEE.platformio\packages\tool-mklittlefs\mklittlefs  exist in this
[image]

xyzroe, 2024-11-12:
Have you tried to use Google?

xyzroe, 2024-11-12:
https://registry.platformio.org/tools/platformio/tool-mklittlefs

xyzroe, 2026-11-15:
Closing: this was a local build environment problem (mklittlefs not found). Build instructions are now in [AGENTS.md](https://github.com/xyzroe/XZG/blob/main/AGENTS.md).

closed 2026-11-15 by xyzroe