xyzroe/XZG #125: Device suddenly entered AP mode?

JzBrevis, 2024-11-11:
I've been using this firmware for around a year but around 20 days ago the LEDs changed, now only the orange LED is continuously lit. It's also opened an unprotected Wifi network that allowed full access to the web config. 

In the web interface, the rightmost icon in the top right group of three icons was not lit stating "unconnected", I assume this means it's saying that the USB is not connected, and that's why it opened the AP?
It still works perfectly fine over USB however, Z2M can still use it and my zigbee devices still function.

Since leaving the AP on would be a massive security risk, as literally anyone could connect to the network and upload malicious firmware and then access my home server which is connected via USB, I disabled some option that said something about selecting a priority to make the web interface available under, and restarted. 

Now the web interface is no longer available, which I guess does fix the security concern, but now I wouldn't know how to access it again?

1. What happened that it entered AP mode while still working perfectly fine over USB?
2. How can I access the web interface again?

I'm sorry, I know this is super vague. I don't have access to the web interface anymore and I didn't take screenshots, and there are no reference screenshots in the docs I can use for better reporting. I thought it would be better to report, maybe someone else has something similar happen?

xyzroe, 2024-11-12:
1. If there isn't active network connection device will start AP. It was a year ago, and still like this. 
2. Find ip of your device and go on it. If there isn't any, you can try to re-flash your gateway using xzg.xyzroe.cc
3. I don't understand why you need gateway if you'r using USB mode.

ArGGu, 2024-12-23:
I have UZG-01 with FW 20240914 connected with POE. The Ethernet connection works just fine, but device says it is disconnected and it starts AP I guess because it detects that Ethernet is disconnect even though it works just fine. I had to connect it to Wireless network to prevent it from creating it's own Wireless network that does not have any authentication enabled on it.

Would be nice to be able to completely disable WiFi or at least set some WPA3-PSK authentication.

etakpte, 2025-04-24:
I have built a version of 20240914  without AP mode if you need.

[XZG.ZIP](https://github.com/user-attachments/files/19898906/XZG.ZIP)

guidojw, 2025-04-28:
I am running XZG on a UZG-01 which is connected to my main network via WiFi, no ethernet.

The UI shows it is in AP mode though and the WiFi network shows up on my phone. How did you guys disable it?

guidojw, 2025-04-28:
Installing the latest beta seems to have resolved it for me.

xyzroe, 2026-11-15:
Closing: as reported above, updating the firmware resolves it, and the AP can be disabled in the network settings. Please also try [20261115](https://github.com/xyzroe/XZG/releases/tag/20261115).

closed 2026-11-15 by xyzroe