xyzroe/XZG #128: Reset Web login

stepman0, 2024-11-15:
**Describe the bug**
I enable username and password for weblogin. Something went wrong (probably on my side), now I cannot login with the password saved in my password manager.

How can I reset the password? I still have access via MQTT.

**Device information**
- Screenshot of Zigbee and Device tab from root page of device
- JSON object returned on [http://xzg.local/api?action=1&param=all](http://xzg.local/api?action=1&param=all) 
 *You **should** remove your credentials first*
  
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

stepman0, 2024-11-30:
No way ???

xyzroe, 2025-01-06:
Re-flashing the device is the only way.

stepman0, 2025-01-06:
Thank you, I did it in the meantime.