xyzroe/XZG #131: ESP32 update wont install on Zigstar UZG-01

storebatfar, 2024-11-29:
I have a Zigstar UZG-01 running ESP32 firmware 20240914. I am trying to install firmware 20241001. Using the UI in the device I start the update, but it just stops during download. Also tried downloading the firmware file and doing a manual update. This stops at 0%.
After several mins. I can refresh the page and the device has rebooted, but not updated the firmware.

Model | UZG-01
Firmware | 20240914
Hardware | ESP32-D0WD-V3

I normally see the update progress from 0-100%, device reboot and all ok.

xyzroe, 2024-12-01:
Use 0914. And don't have any problems.

xyzroe, 2024-12-01:
I don't know at the moment. 
I'm working on an absolutely new version, that is based on clear idf. 
I'll try to support solo boards, but don't know how it will be implemented. 

As I know, there are just some first uzg-01 that use solo boards. About 50-100 pcs.

xyzroe, 2024-12-02:
ESP32-D0WD-V3 is not SOLO

victorhooi, 2024-12-04:
I have a Zigstar UZG-01, which has 

[image]

I can also confirm that I was able to upgrade to the earlier `20240914` build - however, if I try to update to the latest build from October 2024, it just stalls at the `Downloading release 20241001 from Github...` step indefinitely:

[image]

I also tried flashing with the [XZG_20241001.full.bin](https://github.com/xyzroe/XZG/releases/download/20241001/XZG_20241001.full.bin) file from the [XZG Github Releases page](https://github.com/xyzroe/XZG/releases), and that also seems to just hang as well.

xyzroe, 2024-12-05:
If you want to install BETA version you need to use *.ota file form releases. Not full one.

jacobhallgren, 2024-12-10:
[image]
I cannot upgrade my device from 2024-05-19 to 2024-09-14. It just keeps loading.

pullasuti, 2024-12-10:
> If you want to install BETA version you need to use *.ota file form releases. Not full one.

So that's why the flashing fails? The built-in webui flasher is also trying to flash the full .bin file for the beta.

Although flashing the ota file seems to have bricked my D0WD for now.

xyzroe, 2024-12-13:
It's some kind of bug. 
Web UI files must be downloaded automatically while the first boot of XZG, but sometimes it doesn't work. 
So you have two options:
1. Reflash using a USB cable and web flasher. 
2. Upload *.fs.bin manually using PostMan.

yyax, 2025-01-04:
[image]

tangelogh, 2025-01-04:
Same here, stuck on firmware version 20240707..
Wont try to force flash an ota file when it can fail so easily..
Any ETA or even idea on whats the issue?
I can test some things if you want, but kinda sad that I paid almost 50€ for a device with such a conturbated life cycle (started as a UZG, then XZG, then stuck in a random old firmware)..

rickzed, 2025-01-05:
[image]

mihsu81, 2025-01-05:
Why don't you flash it over USB using the web flasher? Had the same issue a few months ago and in 10 minutes I got it flashed with the latest beta and reconfigured.

Bixkitts, 2025-01-05:
`pio run -t upload`
Also results in no web files being uploaded to flash over USB for me (fresh clone of the repo).

xyzroe, 2025-01-05:
Just use version 0914. And don't have any problems.

xyzroe, 2025-01-05:
> what can i do for this [image]
> 

looks like some error in URL. 
Or you're trying on another version than 1001.

xyzroe, 2025-01-05:
> Same here, stuck on firmware version 20240707..
> Wont try to force flash an ota file when it can fail so easily..
> Any ETA or even idea on whats the issue?
> I can test some things if you want, but [...]

In your situation you need to use 0914*.ota.bin file and update manually through the web UI.

xyzroe, 2025-01-05:
> Any ETA or even idea on whats the issue?
> I can test some things if you want, but kinda sad that I paid almost 50€ for a device with such a conturbated life cycle (started as a UZG, then XZG, then stuck in a random old firmware)..

Are you seriously complaining about the price you paid for hardware while using free software that is provided with absolutely no guarantees, on an as-is basis? No one forced you to buy the device if you weren't prepared to accept these terms. Instead of whining, spend your time at least trying to figure things out and testing some solutions yourself. After all, hundreds of users are satisfied and have no issues.

ADMiNZ, 2025-01-05:
[image]

The file is a problem. But that does not help. I also have a favorable connection with the device.

xyzroe, 2025-01-05:
> [image]
> 
> The file is a problem. But that does not help. I also have a favorable connection with the device.

You need to use the *.fs.bin file if there isn't a web UI after the update.

ADMiNZ, 2025-01-06:
> You need to use the *.fs.bin file if there isn't a web UI after the update.

Hello. I have passed by the web interface - the system and overloads international communication or international communication (with ping there are losses in the same rhythm.

xyzroe, 2025-01-06:
Looks like incorrect hardware identification. Use version 0914