xyzroe/XZG #135: Wireguard AllowedIPs/Allowed Subnet

csprr, 2024-12-12:
**Describe the bug**
When trying to add multiple IPs under AllowedIPs, only 1 is accepted

**Device information**
- 
[image]
-
```json
{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": true,
    "ethDhcp": true,
    "ethIp": "::",
    "ethMask": "255.255.255.0",
    "ethGate": "::",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": true,
    "wgLocalIP": "10.99.99.90",
    "wgLocalSubnet": "255.255.255.255",
    "wgLocalPort": <removed>,
    "wgLocalGateway": "10.99.99.99",
    "wgLocalPrivKey": "<removed>",
    "wgEndAddr": "<removed>",
    "wgEndPubKey": "<removed>",
    "wgEndPort": <removed>,
    "wgAllowedIP": "10.99.99.99",
    "wgAllowedMask": "255.255.255.255",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-FA65",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": true,
    "server": "<removed>",
    "port": <removed>,
    "user": "<removed>",
    "pass": "<removed>",
    "topic": "<removed>",
    "updateInt": 30,
    "discovery": true,
    "reconnectInt": 30
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": true,
    "webUser": "<removed>",
    "webPass": "<removed>",
    "fwEnabled": true,
    "fwIp": "<removed>",
    "serialSpeed": 115200,
    "socketPort": <removed>,
    "tempOffset": 20,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 1,
    "hostname": "<removed>",
    "timeZone": "Europe/Amsterdam",
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 1,
    "zbFw": "20240710",
    "updHour": "01:00",
    "updDays": "*",
    "autoIns": false
  }
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'VPN'
2. Insert all details
3. Insert as AllowedIPs: `10.99.99.99,10.99.99.98`
4. Insert as Allowed Subnet: `255.255.255.255,255.255.255.255`
5. Press Save
6. Restart
7. See AllowedIPs being `10.99.99.99` and Subnet `255.255.255.255`, the 2nd one is completely missing

**Expected behavior**
Expected to have access to multiple AllowedIPs, but can only have a single one.

**Screenshots**
- Before save: 
[image]
- After save: 
[image]

**Additional context**
- None

xyzroe, 2024-12-12:
you must read about IP addres and network mask. using both of them allows you to define addresses pull.

xyzroe, 2024-12-12:
ex. https://www.calculator.net/ip-subnet-calculator.html?cclass=any&csubnet=30&cip=10.99.99.96&ctype=ipv4&x=Calculate

csprr, 2024-12-12:
Unfortunately I used the wrong examples and the addresses I am using are in different subnets and can therefore not be specified, since it only accepts 1 address as input.

xyzroe, 2024-12-13:
Ok. So it's impossible to make this using the current WireGuard library. 

I think you could setup IP filtering on your WireGuard server, and on the client just allow 0.0.0.0

csprr, 2024-12-13:
Alright, I should be able to work my way around with that maybe, thank you for the clarification.
The configuration page should be modified though, as this currently states that a comma separated list is accepted; https://xzg.xyzroe.cc/VPN/#wireguard-interface-settings