xyzroe/XZG #136: Web interface not working in latest beta (20241001) - HTTP 404 "File not found"

victorhooi, 2024-12-13:
**Describe the bug**

I have a Zigstar UZG-01 device.

Per the advice on [this issue](https://github.com/xyzroe/XZG/issues/131), I was able to load the latest beta via the .ota file (

However, now when I try to load the web interface for XZG, I get a HTTP 404 error with the text:

`File not found`

**Device information**
- Screenshot of Zigbee and Device tab from root page of device
- JSON object returned on [http://xzg.local/api?action=1&param=all](http://xzg.local/api?action=1&param=all) 
 *You **should** remove your credentials first*

Here is my device info, from the previous working release:

[image]

  
**To Reproduce**
Steps to reproduce the behavior:
1. Load the latest beta release ([XZG_20241001.ota.bin](https://github.com/xyzroe/XZG/releases/download/20241001/XZG_20241001.ota.bin)) onto a Zigstar UZG-01.
2. Let the device reboot itself.
3. Wait for the device to finish booting.
4. Attempt to access the web interface.

**Expected behavior**
The XZG web interface should load successfully.

**Screenshots**

[image]

**Additional context**
Add any other context about the problem here.

xyzroe, 2024-12-13:
https://github.com/xyzroe/XZG/issues/131#issuecomment-2540876772

xyzroe, 2026-11-15:
Closing: after an OTA update of the firmware only, the web files must be updated too (see #131). Flashing the full image or uploading the matching `.fs.bin` fixes "File not found". The new [20261115](https://github.com/xyzroe/XZG/releases/tag/20261115) also downloads the web files automatically.

closed 2026-11-15 by xyzroe