xyzroe/XZG #142: UZG-01 Stuck with Zigbee Firmware 20240909

Asom-Velz, 2025-01-08:
**Describe the bug**
I cant upgrade from Zigbee firmware 20240909

**Device information**
**Zigbee**
Mode	Zigbee-Network
Socket	No connection
Connection time	0 d 00:00:15
Role	Coordinator
Firmware	20240909
Hardware	CC2652P7
IEEE	00:12:4B:00:2A:7B:4C:A2
Flash size	704 KiB

**Device**
Model	UZG-01
Firmware	20241001
Hardware	ESP32-D0WDQ6
Temperature	31.78 °C
Processor	2 cores @ 240 MHz
Heap	        116 / 280 KiB
Memory	4 Mb, internal
NV storage	201 / 630 entries
File system	348 / 1408 KiB
Operating time	0 d 00:01:17
Local time	Wed Jan 8 17:05:14 2025

**Console**
[3111] | [RCP] Chip: CC2652P7
[3112] | [RCP] IEEE: 00:12:4B:00:2A:7B:4C:A2
[3123] | [RCP] Flash size: 704 KB
[11364] | [RCP] FW: Unknown! Check serial speed!
[11479] | [Time] Wed Jan  8 17:24:27 2025
[11480] | [UPD_CHK] cron 27 0 1 * * *
[17701] <- 00 fe
[17715] <- 45 20 28 32 31 34 35 36 29 20 77 fb

**Check Zigbee connect**
[298825] | [RCP] Chip: CC2652P7
[298826] | [RCP] IEEE: 00:12:4B:00:2A:7B:4C:A2
[298836] | [RCP] Flash size: 704 KB
[299816] <- 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 60 00 80 48 02 01 80 f8 89 15 85 2d ad ad ad a5 a5 ad e5 ad ad ed ed cf ef ef ef ef ef ef ff ff

**Check Zigbee version**
[334089] | [RCP] FW: Unknown! Check serial speed!
  
**Screenshots**
[image]
[image]
[image]
[image]

**Additional context**
Trying to upgrade to firmware 
CC1352P7_coordinator_20240316 and it stuck at Update: 64.06%
CC1352P7_coordinator_20240710 and it stuck at Update: 31.00%

Asom-Velz, 2025-01-08:
.

closed 2025-01-08 by Asom-Velz

xyzroe, 2025-01-08:
@Asom-Velz you have closed as completed? 
how did you solve the problem?

Asom-Velz, 2025-01-09:
@xyzroe I finally solve this using Zigstar GW Multi tool with network connection 

I erased the firmware and then I write with Verify the CC1352P7_coordinator_20240316.bin
I repeat this procedure 3 times and the 3rd the Verify reported Checksum Ok!
Between its procedure I removed the network cable (POE) for 1 minute

For your information With my configuration
Home Assistant
    Core 2025.1.1
    Supervisor 2024.12.3
    Operating System 14.1
    Frontend 20250106.0 
+
Zigbee2MQTT Current version: 2.0.0-2

the UZG-01 **only works with CC1352P7_coordinator_20240316.bin**

If I load the CC1352P7_coordinator_20240710.bin The Zigbee2MQTT never connects with the UZG-01

If you want I could open an AnyDesk session for you in order to access my HA and the UZG-01

xyzroe, 2025-01-09:
https://github.com/user-attachments/assets/3856a055-ba47-4ba8-896e-288873fd1f3d

Just checking, and CC1352P7_coordinator_20240710 works as expected. 

But I prefer 20240909 which has some fixes.