xyzroe/XZG #143: Unable to connect router to coordinator

Torx999, 2025-01-10:
I have two of UZG-01 with CC2652P7 . One of them is the coordinator - it works without problems. But the second should be a router to extend the zigbee coverage and I can't get it connected to the coordinator.

The router has the last XZG ESP32 Firmware 20241001 (the coordinator has this version too) and I tried it with both available router zigbee firmwares: 20231201 and 20dBm_20240909 - both without success. for tests there was about 50cm between both devices, so the weak signal is not a problem. factory resets, router reconnect button (in tools/web UI) - no success. 
Coordinator has some zigbee devices connected, so i don't think the problem is in it.
Zigbee infrastructure is managed from iobroker with zigbee adapter.

**To Reproduce**
Steps to reproduce the behavior:

1. flash router firmware for router device
2. allow zigbee joins for coordinator
3. power on the router, or press "router reconnect" button in web ui

**Expected behavior**
successful join of the router to existing zigbee network

xyzroe, 2025-01-10:
I don't think the described problem is very related to the XZG firmware

Torx999, 2025-01-14:
The problem was - too low power from poe injector. It's still unclear why it is enough to act as coordinator, but not enough to act as a router. But after replacing an injector the router also connect like a charm.

closed 2025-01-14 by Torx999