xyzroe/XZG #144: Can't Build And Upload Properly From Terminal

Bixkitts, 2025-01-14:
**Describe the bug**
When I use the command `pio run` from the terminal, binaries don't get emitted to
./bin/, except for XZG.fs.bin for some reason.
According to the console, ./.pio/build/prod-dual/firmware.bin is flashed to the chip
(with `pio run -t upload`. Default env is prod-dual.).

When I make changes to the HTML, then build and upload, changes aren't
reflected on the actual hardware so I'm led to believe the web files
aren't making their way into the binary properly.

Before I spend more time reading through the build system, and disassembling binaries,
am I doing something silly, or is this a bug?

Thank you!
  
**To Reproduce**
1. git clone
2. Make arbitrary changes in src/websrc/html/
3. `pio run -t upload`
4. Arbitrary changes are not reflected in actual webserver on hardware

**Expected behavior**
On older versions (A few months ago) I had almost no issues using this workflow
of making edits and uploading with platformio from the terminal.

xyzroe, 2025-01-14:
https://github.com/xyzroe/XZG/tree/main/tools%2Fwebfilesbuilder

Here you can find answers

Bixkitts, 2025-01-14:
I did read through the webfile processing steps before asking here, thank you!
Everything in webfilesbuilder/ appears to execute correctly and I get a filesystem binary (./bin/XZGfs.bin is produced).

However, something seems to go wrong afterwards.
I think `after_build()` in build.py doesn't execute and my binary gets uploaded before `merge_bin_esp.py`...

This stuff should "just work" like it did before- what are you guys doing to have it build and upload without issue?

Bixkitts, 2025-01-14:
Here is the stdout from the compilation and upload!
As you can see, I don't get log output from merge_bin_esp.py and it says "Uploading .pio/build/prod-dual/firmware.bin" on line 123...
[log.txt](https://github.com/user-attachments/files/18409108/log.txt)

Bixkitts, 2025-01-15:
Managed to merge together a XZG.full.bin, "completed successfully" with this command:
`python tools/build/merge_bin_esp.py --output_folder ./bin --output_name XZG.full.bin --bin_path .pio/build/prod-dual/bootloader.bin .pio/build/prod-dual/partitions.bin .pio/build/prod-dual/firmware.bin bin/XZG.fs.bin --bin_address 0x1000 0x8000 0x10000 0x290000`
Instead of:
`python tools/build/merge_bin_esp.py --output_folder ./bin --output_name XZG.full.bin --bin_path bin/bootloader_dio_40m.bin bin/partitions.bin bin/firmware.bin bin/XZG.fs.bin --bin_address 0x1000 0x8000 0x10000 0x290000`

Bixkitts, 2025-01-15:
Ah okay so firmware.bin/OTA (what gets uploaded with pio run -t upload) is _just_ the firmware (without web files or bootloader)... when I worked on this last year filesystem was uploaded together with firmware.

xyzroe, 2025-01-15:
On my setup all works as expected. Full fw (with bootloader and file system) loads to esp. 
what's going on with your setup, - only you can figure it out.

Bixkitts, 2025-01-15:
Skill issue?
Nice!
If I patch something that maybe helps other people run the build, I'll send it upstream.
Thanks for the tip with running the merge manually and letting me know the filesystem
_is_ supposed to be in the uploaded binary, closing.

closed 2025-01-15 by Bixkitts