xyzroe/XZG #146: Update to 20240914 neither via GUI nor Web installer working

Alex71w, 2025-01-27:
**Describe the bug**
Trying to install the ESP-Firmware Ver 20240914 (currently installed: 20240510)
Several tries via Web-Interface, directly via "Git ESP32-Update" as well as via downloaded SW and then "File ESP32-Update". The progress bar is running and running and running... this bug is already described in other issues.

I then tried it via the Web-installer. I downloaded the driver CH341SER (for CH341 and CH340) and installed it. I chose Firmware Ver 20240914, and hit the button "install", selecting COM-Port3. The device is found, and the installation process starts. It stops after some time, saying: "Cannot install. Try resetting the devise or press the Boot-Button while installing".

I tried the second one, and long-pressed the "boot"-button on my device, but same error message.
As I do not know how to reset: I pressed the RST-button on my device, but nothing changed.

Lucky me the device is still functioning.

**Device information**
- Screenshot of Zigbee and Device tab from root page of device
[image]

- JSON object returned on [http://xzg.local/api?action=1&param=all]
{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "ethEnbl": true,
    "ethDhcp": true,
    "ethIp": "::",
    "ethMask": "255.255.255.0",
    "ethGate": "::",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "::",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-A0B3",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": false,
    "server": "",
    "port": 1883,
    "user": "mqttuser",
    "pass": "",
    "topic": "ZigStarGW-21E7",
    "updateInt": 60,
    "discovery": false,
    "reconnectInt": 0
  },
  "system-config": {
    "keepWeb": false,
    "disableWeb": false,
    "webAuth": false,
    "webUser": "",
    "webPass": "",
    "fwEnabled": false,
    "fwIp": "0.0.0.0",
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 3,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 1,
    "hostname": "ZigStarGW",
    "timeZone": "Europe/Kiev",
    "ntpServ1": "pool.ntp.org",
    "ntpServ2": "time.google.com",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0
  }
}
  
**To Reproduce**
See Bug-Description

**Expected behavior**
Successfull update of ESP-Firmware either via Web-Interface or via Installer.

**Screenshots**
n/a

**Additional context**
n/a

xyzroe, 2025-01-28:
What file you are using when trying to make file based update?

According to unworkable web installer - for all users it works great. You should double check COM port, drivers, also that any other software doesn't use this port at the same time. 
And on WT32 you need to press boot button before powering up and release it only after firmware process has started.

Alex71w, 2025-01-28:
Thank you. I will try this in the next days and hopefully it solves the
problem!