xyzroe/XZG #149: Trouble Integrating Zigstar UZG-01 in Home Assistant – Unknown Error

robbionline, 2025-02-09:
I've recently reconfigured my Home Assistant setup and wanted to share my experience:

1. I switched my **SkyConnect** to act as my **Thread Border Router** after reading about the current issues with multiprotocol support.
2. I tested a separate Zigbee stick (**SONOFF Zigbee Gateway, ZBDongle-E 3.0 USB Dongle Plus, EFR32MG21 + CH9102F/CP2102N**), but I returned it due to issues with IKEA devices.
3. Now, I'm trying the **Zigstar UZG-01**, hoping that its **Ethernet connection** will provide more flexibility.

### The Issue

Unfortunately, when I attempt to add the UZG-01 (both manually and via automatic discovery), I get an **"Unknown Error"**:
[image]

Logs from Home Assistant while trying to add the UZG-01:
```
2025-02-01 00:50:27.754 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=0)
2025-02-01 00:50:27.762 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.NVLength.Rsp(Length=20)
2025-02-01 00:50:27.763 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=0, Offset=0, Length=20)
2025-02-01 00:50:27.773 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2025-02-01 00:50:27.773 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0000] = b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00'
2025-02-01 00:50:27.774 DEBUG (MainThread) [zigpy_znp.api] Detected Z-Stack 3.3
2025-02-01 00:50:27.774 DEBUG (MainThread) [zigpy_znp.api] Connected to socket://192.168.178.130:6638
2025-02-01 00:50:27.774 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.ResetReq.Req(Type=<ResetType.Soft: 1>)
2025-02-01 00:50:27.774 DEBUG (MainThread) [zigpy_znp.api] Request has no response, not waiting for one.
2025-02-01 00:50:27.775 DEBUG (MainThread) [zigpy.serial] Waiting for serial port to close
2025-02-01 00:50:27.775 DEBUG (MainThread) [zigpy.serial] Connection lost: None
```
Debug Log from the UZG-01:
```
[784133] | Connected client 1 from 192.168.178.46
[784586] -> fe 00 21 01 20
[784590] <- fe 02 61 01 59 06 3d
[784593] -> fe 01 27 49 00 6f
[784602] <- fe 24 67 49 fe ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0b
[784606] -> fe 05 21 32 01 04 00 00 00 13
[784611] <- fe 04 61 32 14 00 00 00 43
[784613] -> fe 08 21 33 01 04 00 00 00 00 00 14 0b
[784622] <- fe 16 61 33 00 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 af
[784624] -> fe 01 41 00 01 41
[786151] <- fe 06 41 80 00 02 01 02 07 01 c0
```
Zigbee Firmware:
```
Role	Coordinator
Firmware	20240710
```
Hardware Details:
```
Model	UZG-01
Firmware	20241001
Hardware	ESP32-D0WD-V3
```

xyzroe, 2025-02-12:
https://github.com/user-attachments/assets/98a723e2-dda6-4525-af22-8db0d0108872

xyzroe, 2025-02-12:
I just tried adding XZG to ZHA. 
On the above post you can find a video demonstration. 
I did not encounter the problems you describe.

I think you should use 0914 but not 1001 if there are any problems with that.

closed 2025-03-21 by xyzroe