xyzroe/XZG #150: OpenThread router configuration - how?

q20, 2025-02-21:
I am attempting to configure a UZG-01 to provide OpenThread support to Home Assistant.

I have changed operating mode here:

[image]

…which results in the system downloading and flashing `CC1352P7_ot_rcp_2025_2_1_20250203.bin`. Once done, the system shows 20250203 firmware installed:

[image]

Um, now what? Zigbee mode (above) still shows Zigbee-Network, not OpenThread.
How do I set the device to run an OpenThread configuration and, once that's done, connect to Home Assistant?
I have tried connecting to the IP:PORT of the UZG-01, but:

[image]

…this fails:

[image]

Any ideas?
Thanks!

xyzroe, 2025-02-21:
tcp:// or socket://

xyzroe, 2025-02-21:
you know there's no turning back, right?

q20, 2025-02-21:
> you know there's no turning back, right?

Yup. I am indeed.   👍
Unfortunately, neither tcp: nor socket: work:

[image]

Zigbee firmware now shows:

[image]

What could be happening here?

PolarWooolf, 2025-02-24:
> Um, now what? Zigbee mode (above) still shows Zigbee-Network, not OpenThread.
It's a bug. Your device is in OTBR mode.

You need to install "OpenThread Border Router" HASSIO addon and fill network address into its configuration like that: 

[image]

q20, 2025-02-24:
Thank you! That was the tip I needed. My device now shows up as an OpenThread border router:

[image]

Case closed.   👍

closed 2025-02-24 by q20

mandorakatiki, 2025-03-04:
> you know there's no turning back, right?

Why?

xyzroe, 2025-03-05:
> > you know there's no turning back, right?
> 
> Why?

cause after flashing, the chip does not respond to standard commands to enter flash mode.