xyzroe/XZG #152: UZG-01 crashes with Zigbee2Mqtt

juergen852, 2025-02-28:
**Describe the bug**
- Starting zigbee2mqtt as docker compose.
- Z2M and Adapter are in same LAN
- error:     z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

docker compose up
```
[+] Running 2/2
 ✔ Container zigbee2mqtt         Created                                                                           0.0s 
 ✔ Container zigbee2mqtt-mqtt-1  Created                                                                           0.0s 
Attaching to zigbee2mqtt, mqtt-1
zigbee2mqtt  | Using '/app/data' as data directory
mqtt-1       | 1740744419: mosquitto version 2.0.20 starting
mqtt-1       | 1740744419: Config loaded from /mosquitto-no-auth.conf.
mqtt-1       | 1740744419: Opening ipv4 listen socket on port 1883.
mqtt-1       | 1740744419: Opening ipv6 listen socket on port 1883.
mqtt-1       | 1740744419: mosquitto version 2.0.20 running
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | [2025-02-28 13:07:00] info:      z2m: Logging to console, file (filename: log.log)
zigbee2mqtt  | [2025-02-28 13:07:00] info:      z2m: Starting Zigbee2MQTT version 2.1.1-dev (commit #102857a7fec60e3f65887d92b6fe7a2d81add110)
zigbee2mqtt  | [2025-02-28 13:07:00] info:      z2m: Starting zigbee-herdsman (3.2.7)
zigbee2mqtt  | [2025-02-28 13:07:00] info:      zh:zstack:znp: Opening TCP socket with 10.8.1.171:6638
zigbee2mqtt  | [2025-02-28 13:07:00] info:      zh:zstack:znp: Socket connected
zigbee2mqtt  | [2025-02-28 13:07:00] info:      zh:zstack:znp: Socket ready
zigbee2mqtt  | [2025-02-28 13:07:00] info:      zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
zigbee2mqtt  | [2025-02-28 13:07:01] info:      zh:zstack:znp: Skip bootloader for CC2652/CC1352
zigbee2mqtt  | [2025-02-28 13:07:19] error:     z2m: Error while starting zigbee-herdsman
zigbee2mqtt  | [2025-02-28 13:07:19] error:     z2m: Failed to start zigbee-herdsman
zigbee2mqtt  | [2025-02-28 13:07:19] error:     z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
zigbee2mqtt  | [2025-02-28 13:07:19] error:     z2m: Exiting...
zigbee2mqtt  | [2025-02-28 13:07:19] error:     z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
zigbee2mqtt  |     at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
zigbee2mqtt  |     at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:69:27)
zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:142:13)
zigbee2mqtt  |     at start (/app/index.js:161:5)
zigbee2mqtt exited with code 0

```

Adapter-Log:----------------------------------------------------
```
[60233837] | Connected client 1 from 10.8.1.184 <-- IP of docker Z2M
[60233838] -> ef
[60234838] -> fe 00 21 01 20
[60240844] -> fe 00 21 01 20
[60246850] -> fe 00 21 01 20
```

**Device information**
[image]

```
{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "0.0.0.0",
    "wifiMask": "255.255.255.0",
    "wifiGate": "0.0.0.0",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": true,
    "ethDhcp": true,
    "ethIp": "0.0.0.0",
    "ethMask": "255.255.255.0",
    "ethGate": "0.0.0.0",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "0.0.0.0",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-7420",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": false,
    "server": "",
    "port": 1883,
    "user": "",
    "pass": "",
    "topic": "XZG-7420",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 15
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": false,
    "webUser": "",
    "webPass": "",
    "fwEnabled": false,
    "fwIp": "0.0.0.0",
    "fwMask": "0.0.0.0",
    "serialSpeed": 115200,
    "socketPort": 6638,
    "tempOffset": 9,
    "disableLedUSB": false,
    "disableLedPwr": false,
    "refreshLogs": 2,
    "hostname": "ZIGBEE01",
    "timeZone": "Europe/Berlin",
    "ntpServ1": "de.pool.ntp.org",
    "ntpServ2": "pool.ntp.org",
    "nightMode": false,
    "startHour": "23:00",
    "endHour": "07:00",
    "workMode": 0,
    "zbRole": 1,
    "zbFw": "20240710",
    "updHour": "01:00",
    "updDays": "*",
    "autoIns": false
  }
}
```

**To Reproduce**
Docker Compose----------------- docker-compose.yaml
```
services:
    zigbee2mqtt:
        container_name: zigbee2mqtt
        restart: unless-stopped
        image: koenkk/zigbee2mqtt:latest-dev
        volumes:
            - ./zigbee2mqtt-data:/app/data
            - /run/udev:/run/udev:ro
        ports:
            - '8080:8080'
            #- 10.8.1.184:8080:8080
        environment:
            - TZ=Europe/Berlin
        #devices:
        #    - /dev/ttyUSB0:/dev/ttyUSB0
    mqtt:
        image: eclipse-mosquitto:2.0
        restart: unless-stopped
        volumes:
            - './mosquitto-data:/mosquitto'
        ports:
            - '1883:1883'
            - '9011:9011'
        command: 'mosquitto -c /mosquitto-no-auth.conf'

```
Adapter Config------------------- zigbee2mqtt-data/configuration.yaml

```
homeassistant:
  enabled: false
frontend:
  enabled: true
  port: 8080
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://10.8.1.176:1883
serial:
  port: tcp://10.8.1.171:6638
  adapter: zstack
  baudrate: 115200
  disable_led: false
advanced:
  transmit_power: 20
  channel: 11

```

**Expected behavior**
I would expect zigbee2mqtt to start up

xyzroe, 2025-03-02:
Your screenshot shows that your Zigbee module doesn't have any current firmware (If the version number of a Zigbee module is displayed in italics, it is the latest version that has ever been detected on that module.) 
Most likely you were “playing” with the firmware and something went wrong.

closed 2025-03-02 by xyzroe

juergen852, 2025-03-05:
**Thanks** for the "italics" hint, I did not know that.

After updating everythin again, Z2M starts without Problems.