xyzroe/XZG #155: xzg|usg-01 freeze on working

vpaulos, 2025-03-04:
New clean installation of a new device. 
Zigbee && mqtt were configured. 

I noticed that only one new device appeared in mqtt server - the controller itself, and no new sub devices (which are connected to the controller) appeared in mqtt, although they appeared in zha.

After some time I noticed that the devices do not respond to commands from the web interface zha and only rebooting the device (usg-01) helps.

Attempt to enter the web interface after the problem starts always ends unsuccessfully. The web interface is either unavailable or the "loading ball" is spinning and that's it. 
You can wait for 10 minutes and only a hardware restart helps.

[image]

[image]

vpaulos, 2025-03-04:
after full reset.
wait for freezing

[image]

vpaulos, 2025-03-04:
[1140] | [ZB] Chip: CC2652P7
[1141] | [ZB] IEEE: 00:12:4B:00:2E:10:4C:15
[1152] | [ZB] Flash size: 704 KB
[3925] | [Time] Tue Mar  4 06:57:18 2025
[3925] | [UPD_CHK] cron 3 0 1 * * *
[4107] | [DNS] Saved ETH - 192.168.220.254
[4153] | [ZB] FW: 20240710
[13081] | Found ESP new fw: 20241001.full
[75656] | ZB RST begin
[76205] | ZB restart was done
[77757] <- fe 06 41 80 00 02 01 02 07 01 c0
[78910] | Router rejoin begin
[79660] | Router in join mode!

vpaulos, 2025-03-04:
freeze always was after trying to update to lates version.. in the update proccess, it cant download frmware and freeze allll... (((

[image]

xyzroe, 2025-03-04:
Unfortunately, there's a lot of text but not much information.

xyzroe, 2025-03-04:
> [78910] | Router rejoin begin
> [79660] | Router in join mode!

this logs shows that you have enabled router joining mode, that is unnecessary while in coordinator mode.

xyzroe, 2025-03-04:
> appeared in mqtt server - the controller itself, and no new sub devices (which are connected to the controller) appeared in mqtt, although they appeared in zha.

You should learn how is ZHA working.... 
(Devices shouldn't appear in MQTT)

> After some time I noticed that the devices do not respond to commands from the web interface zha and only rebooting the device (usg-01) helps.

That shows that there're some problems with Zigbee network. Not related to XZG firmware itself.

closed 2025-03-21 by xyzroe