xyzroe/XZG #159: UZG-01 bricked

Aussie-Luftikus, 2025-03-26:
Hello XZG Team
 
I bought a Zigstar UZG-01 (Model UZG-POE-0.5) a couple of months ago. It worked ok, but I recently updated the firmware because a popup in the web interface informed me of a new update.
 
There was no issue during the update process that I am aware of and the status page after the update indicated version XZG_20241001. The popup window didn’t mention anything about beta firmware, and had I known, I probably wouldn’t have installed this beta version. After the update it was immediately clear that this firmware is a disaster. First, I noticed that the Ethernet port was gone from the web UI and looking at the physical port itself, there was no light flashing as usual. The next thing I noticed was that the status window does no longer show a version for the Zigbee firmware. I then tried to flash the Zigbee firmware and that is where I got stuck. I meanwhile reverted to XZG_20240914 and my ethernet port came back. However, it didn’t fix the issue with the Zigbee firmware. I have tried many different Zigbee firmware versions, but none of them installed. The latest I tried is CC1352P7_coordinator_20240710.hex.
 
I have spent many days trying to find information that helps unbrick the UZG-01, but nothing was successful. I have tried to flash via network and via USB and I always see the error message “Timeout waiting for ACK/NACK after ‘Synch (0x55 0x55)’”. The button doesn’t seem to have a function and the lights never change, it always shows a dim blue light (not flashing).
 
It is incredibly frustrating and disappointing to use this device. Most information I could find on the web was outdated and not relevant and all the many firmware versions are a total disaster to navigate. If I would account for the time I spend trying to get this thing to work, it would have cost me thousands of dollars.
 
Contacting you guys is my last attempt to get this thing working bevor I throw it in the bin. Maybe you have a way to flash the Zigbee firmware that I am not yet aware of and would appreciate some help in this regard.
 
Regards
Dirk Lessner

[image]

xyzroe, 2025-03-26:
Dear Dirk,

Thank you for reaching out. We understand your frustration, but we would like to clarify a few points:

1. XZG is an open-source firmware provided without any obligations or guarantees. It is not affiliated with Zigstar UZG in any way.

2. XZG_20241001 is a beta version, and all relevant details were clearly stated in the release notes.

3. Flashing an incorrect firmware on the Zigbee chip likely caused a wrong bootloader to be installed. To restore the correct bootloader, you will need an external J-Link programmer. For future updates, make sure to only use firmware with the "P7" suffix.

4. We recommend contacting your seller, as they might be able to assist you with recovery.

Best regards,
XZG Team

Aussie-Luftikus, 2025-04-01:
XZG Team

 

Thank you for responding.

 

I understand that you recognise my frustration but cannot offer help and rather send me off to the seller.

 

Let me comment on your four items in your response:

 
I consider it an affiliation when ZigStar points to XZG firmware as the preferred option.
I know that XZG_20241001 is beta. And as such it should not be promoted to the top of the firmware list. In fact, it should not be shown at all by default and only show up when selected, e.g. by selecting “Show Beta Versions”. The latest stable version is then hidden somewhere down the list and easily overlooked.
I agree. However, I did not intentionally install a wrong firmware version. The wrong firmware version is selected as default and cannot be changed. Therefore, I could argue that the build in update functions are suicide functions, implemented by the developer. For example, the version installed under Zigbee/Device Role/Coordinator is CC1352P2_CC2652P_launchpad_coordinator_20240710.bin and not the version with “P7” suffix you mentioned. Or if I try that under Tools/Firmware/ESP32/Install latest, it will do just that – install the Beta with no option to select a different version. If I click on Show available underneath, I get the list with two beta versions on the top of the list, where they should not show up there at all if they are so dangerous. Furthermore, if I install the latest stable version (not beta), I get constantly bothered with a popup window that there is an update available. And if I click on install now, guess what – it installs the beta version. Finally, your comment under point 3 that I need to make sure to only use firmware with “P7” suffix is kind of ridiculous considering that XZG firmware is full of these suicide update functions that, in particular with people not familiar with the ESP32 eco system, don’t leave an option to choose a firmware.
Going back to ZigStar to ask them for help with a problem that was caused by XZG firmware is a bit over the top, don’t you think?
 

Thanks for all your help and hard work, much appreciated.

 

Regards

Dirk Lessner

Reply to: xyzroe/XZG ***@***.***>
Date: Wednesday 26 March 2025 at 18:19
To: xyzroe/XZG ***@***.***>
Cc: Aussie-Luftikus ***@***.***>, Author ***@***.***>

 

"Organization", "name": "GitHub", "url": "https://github.com" } } ]

xyzroe, 2025-04-04:
@Aussie-Luftikus once again

XZG is an open-source firmware provided without any obligations or guarantees

it's FREE project without ANY GUARANTEES

xyzroe, 2025-04-04:
Some more details:

> I understand that you recognise my frustration but cannot offer help and rather send me off to the seller.
1. You are **wrong**. I see your unwillingness to understand the problem and look for ways to solve it. But then I see well your desire to teach someone how to do it.

> I consider it an affiliation when ZigStar points to XZG firmware as the preferred option.
2. And here you are **wrong**. This is nothing more than the choice and desire of the manufacturer of the gateway.

> And as such it should not be promoted to the top of the firmware list. In fact, it should not be shown at all by default and only show up when selected, e.g. by selecting “Show Beta Versions”. The latest stable version is then hidden somewhere down the list and easily overlooked.
3. I am always glad of new commits, you can realize what you describe and send commit.

> However, I did not intentionally install a wrong firmware version.
4. You made this sequence of rash actions. Without studying instructions.

> The wrong firmware version is selected as default and cannot be changed. Therefore, I could argue that the build in update functions are suicide functions, implemented by the developer. For example, [...]
5. **These are your wrong guesses.** As in fact - the Zigbee chip model is definite (displayed on the main page), the correct version of the firmware will be installed. But if you removed the bootloader from your Zigbee chip (probably by manual installation of the wrong file), then the chip model cannot be defined. 

https://github.com/user-attachments/assets/384dda99-eb1f-4fa5-b699-0fb333527b7c

> Or if I try that under Tools/Firmware/ESP32/Install latest, it will do just that – install the Beta with no option to select a different version.
6. These are your **wrong** guesses. As in fact - it is impossible to install beta from that button. To prevent accidental installation.

https://github.com/user-attachments/assets/e6d06566-b986-40cb-b2da-3877a0a7d08f

> If I click on Show available underneath, I get the list with two beta versions on the top of the list, where they should not show up there at all if they are so dangerous. 
7. Well, **yes**. That's how beta versions **always** work. Those who are ready to test them install them and give feedback.

> Furthermore, if I install the latest stable version (not beta), I get constantly bothered with a popup window that there is an update available.
8. How often do you need to go into the web interface that you call it **bothered**?

> And if I click on install now, guess what – it installs the beta version.
9. And here you are **wrong**, again. You'll get a popup with a list of versions. 

https://github.com/user-attachments/assets/6d922eb9-3287-485a-9c34-3b5c35018708

> Finally, your comment under point 3 that I need to make sure to only use firmware with “P7” suffix is kind of ridiculous considering that XZG firmware is full of these suicide update functions that, [...]
10. When everything is OK with your device (and this is the device you usually get after purchase), then the ZigBee chip model is correctly determined, and as I wrote above, the correct firmware version will always be installed for it. and only if YOU erased or damaged the bootloader, the model cannot be determined - you need to carefully watch what you install through the menu Tools - Firmware - ZigBee. 

> Going back to ZigStar to ask them for help with a problem that was caused by XZG firmware is a bit over the top, don’t you think?
11. The problem is not caused by the firmware. the problem is caused by **your** thoughtless and probably chaotic actions.

To sum it up: I got the impression that you either don't want to understand what you're doing at all, or you're deliberately trying to distort the facts and twist the situation.
In addition, you haven't even tried to structure your actions and the problems you've encountered in a format that's easy for others to read.

I repeat: this is a free project, with open source, without any guarantees or obligations.
The firmware works with different models of coordinators. Contributions can and are made by completely different people. And a much larger number of people use this firmware and are very happy that it exists in this form.

If you personally don't like something about the firmware - just don't use it.

If you have problems with the device - contact the seller or manufacturer.

closed 2025-04-04 by xyzroe