xyzroe/XZG #160: UZB-01 Lan coordinator how to change IEEE Adress?

silversurfa, 2025-04-13:
**Is your feature request related to a problem? Please describe.**
Hello! 
I use the UZB-01 and want to migrate Zigbee2MQTT from my old Sonoff Zigbee Dongle without re-pairing all Zigbee devices

[**Describe]([url](

[image]

)) the solution you'd like**
Before i ordered the UZB-01 i googled a lot and found that the IEEE Adress of the new coordinator must be changed to the old Sonoff IEE to make that work. It was shown in the web interface in a submenu.
Is there a chance to do it within my webinterface?

See the status page in the screenshot of my Webinterface attached.

Thank you for helping me out

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

xyzroe, 2025-04-14:
To migrate you don't need to change IEEE, that's wrong. 
That's impossible to change main IEEE address in TI2652* chips.

xyzroe, 2025-04-14:
But I see some problem on you screen shot. Version of Zigbee FW shown italic - that's mean there isn't real answer from Zigbee module during boot up. Seems you need to reflash Zigbee module, but be careful.

christopheVia, 2026-03-02:
Thanks. I'm trying but without any success for now. XZG-MT cannot detect my device (attached to the computer with usb). I don't see how to correctly configure the tool for this coordinator (usb connection).

> You can use https://mt.xyzroe.cc/ to change the IEEE https://github.com/xyzroe/XZG-MT