xyzroe/XZG #161: May I ask how to change the model recognized by xzg firmware？

gx19970920, 2025-04-25:
I am using UZG-01 hardware, but the firmware page shows Olizig. Can this be modified?

[image]

xyzroe, 2025-04-25:
There is some problem with correctly determining the model. 
I recommend to use 0914 version. 
There is also a possibility to change config (including model name) by hands, in 0914.

gx19970920, 2025-04-26:
Thank you for the recovery. I have manually downloaded "XZG_20240914. full. bin" and upgraded it through ESP32 manual flashing on the web page, but it keeps prompting "Waiting for device...". How can I handle this

gx19970920, 2025-04-27:
Thank you very much for your reply!