xyzroe/XZG #164: Device Bricked?

mpl1337, 2025-05-14:
Hello, i tryed to Update the Zigbee coordinator Firmware to Beta_20250321

Iv got the Update Popup but after 10 minutes no result and still the same Message

[image]

After Reboot was my Zigbee Coordinator deleted and now i cant install a new/other one

[image]

Try to Flash 20240710 Coordinator:

`[5677] | [RCP] Chip: CC2652P2_
[5677] | [RCP] IEEE: 00:12:4B:00:24:CB:F7:07
[5688] | [RCP] Flash size: 352 KB
[13929] | [RCP] FW: Unknown! Check serial speed!
[14546] | [Time] Wed May 14 20:43:34 2025
[14548] | [UPD_CHK] cron 55 0 1 * * *
[16550] | Connected client 1 from 192.168.178.148
[17003] -> fe 00 21 01 20
[17564] -> ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef
[20007] -> fe 00 21 01 20
[21076] -> fe 00 21 01 20
[22020] -> fe 00 21 01 20
[23010] -> fe 00 21 01 20
[24012] -> fe 00 21 01 20
[25013] -> fe 00 21 01 20
[26014] -> fe 00 21 01 20
[33222] | Connected client 1 from 192.168.178.148
[33670] -> fe 00 21 01 20
[34198] -> ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef
[36683] -> fe 00 21 01 20
[37678] -> fe 00 21 01 20
[38682] -> fe 00 21 01 20
[39679] -> fe 00 21 01 20
[40680] -> fe 00 21 01 20
[41682] -> fe 00 21 01 20
[42682] -> fe 00 21 01 20
[54997] | Connected client 1 from 192.168.178.148
[55452] -> fe 00 21 01 20
[55972] -> ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef
[58450] -> fe 00 21 01 20
[59451] -> fe 00 21 01 20
[60454] -> fe 00 21 01 20
[61454] -> fe 00 21 01 20
[62455] -> fe 00 21 01 20
[63456] -> fe 00 21 01 20
[64465] -> fe 00 21 01 20
[70369] | Start Zigbee flashing
[70369] | ZB flash https://raw.githubusercontent.com/xyzroe/XZG/zb_fws/ti/coordinator/CC1352P2_CC2652P_launchpad_coordinator_20240710.bin @ 115200
[71443] | Erase completed!
[71446] | Failed to download file, HTTP code: -1

[71959] | Erase completed!
[71961] | Failed to download file, HTTP code: -1

[72474] | Erase completed!
[72477] | Failed to download file, HTTP code: -1

[72990] | Erase completed!
[72992] | Failed to download file, HTTP code: -1

[73505] | Erase completed!
[73507] | Failed to download file, HTTP code: -1

[74020] | Erase completed!
[74022] | Failed to download file, HTTP code: -1

[74535] | Erase completed!
[74538] | Failed to download file, HTTP code: -1

[75588] | Failed to flash Zigbee
[80140] | [Time] Wed May 14 20:44:39 2025
[80141] | [UPD_CHK] cron 11 0 1 * * *
[86640] | Connected client 1 from 192.168.178.148
[87049] -> fe 00 21 01 20
[87579] -> ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef
[90053] -> fe 00 21 01 20
[91053] -> fe 00 21 01 20
[92055] -> fe 00 21 01 20
[93056] -> fe 00 21 01 20
[94061] -> fe 00 21 01 20
[95059] -> fe 00 21 01 20
[96064] -> fe 00 21 01 20
[137907] | Connected client 1 from 192.168.178.148
[138356] -> fe 00 21 01 20
[138883] -> ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef ef
[141365] -> fe 00 21 01 20
[142370] -> fe 00 21 01 20
[143414] -> fe 00 21 01 20
[144420] -> fe 00 21 01 20
[145412] -> fe 00 21 01 20
[146421] -> fe 00 21 01 20
[147415] -> fe 00 21 01 20`

xyzroe, 2025-05-15:
First you need to make ESP firmware to work properly over Ethernet.

Maybe wrong model detection? Or you really have Olimex based ZigStar? 
If there is any error in model detection, you can try to install earlier version (202405*), make factory reset, check that model was determined right, and after all update ESP32 to 20240914 or *1001.

In the end you can re-flash your Zigbee chip using web UI. 
Or you can try to use any CC2652 flasher that supports TCP connection, but enable BSL manually using debug page, right before the start of flashing process.

ffries, 2025-05-31:
When updating Zigbee firmware, make sure to disable zigbee connections to your coordinator while upgrading otherwise you may brick your device and you need to reflash.

ciddi89, 2025-07-12:
Hey @mpl1337, did you manage to find a solution for your issue?

I’m facing the same problem with an Ethernet Zigbee coordinator from cod.m. I attempted to update the firmware, but the process always fails — usually breaking off around 30%. I tried several different firmware versions, but none worked reliably.

Interestingly, when I use ZigstarGW-MT, the update runs through to 100% and says “firmware update OK”. However, my Zigbee devices remain unresponsive, and Zigbee2MQTT doesn’t start or becomes unreachable, even after restarting all systems (including the coordinator and host).

Any ideas or updates would be appreciated!

The debug console from the coordinator says:

```
[1136] | [ZB] Chip: CC2652P7
[1136] | [ZB] IEEE: 00:12:4B:00:2E:13:3A:83
[1147] | [ZB] Flash size: 704 KB
[4110] | [Time] Sat Jul 12 16:59:22 2025
[4111] | [UPD_CHK] cron 1 0 1 * * *
[6984] | [DNS] Saved ETH - 172.20.50.6
[15889] | [ZB] FW: Unknown! Check serial speed!
[22429] | Connected client 1 from 172.20.30.200
```

xyzroe, 2026-11-15:
Closing as a duplicate of #70 (Zigbee flashing stalls while clients are connected). Stop ZHA/Z2M (or Home Assistant completely) before flashing, then re-flash the coordinator firmware from the web UI.

closed 2026-11-15 by xyzroe