xyzroe/XZG #165: MQTT - increase password limit

fliespl, 2025-05-24:
**Describe the bug**
There is probably a limit on number of chars used by mqtt password field. After saving my pass is being cut off, making it not working at all.

My current password:
vieF8ooquoopaegh3iepohchoh2eec8jaeChashoh5shejuChafaeT7zashoongu

xyzroe, 2025-05-25:
Why don't you use a shorter password?

fliespl, 2025-05-25:
@xyzroe or course I can and I did that to set it up adhoc, but how is that relevant to the bug reported? :)

Mqtt protocol specs allow password to be a string with a maximum of 65535 bytes, so I am reporting that this condition in xzg fails.

Also current password recommendations from NIST mention using longer passwords instead of complex ones.

xyzroe, 2025-05-25:
https://github.com/xyzroe/XZG/blob/main/src%2Fconfig.h#L192

I think 50 symbols is quite enough. (49 actually)
But you can adjust by yourself.

fliespl, 2025-05-25:
@xyzroe I am not forcing anything :) 

Like I said - I only mentioned this bug, cause someone else might end up with this problem.

And in my case it involved 30 minutes of troubleshooting connectivity settings for connectivity issues. Password was cut off "silently" (no error / no warning). If I would get an error or something I wouldn't even bother you :)

nabeelr, 2025-08-29:
I'm having a similar yet opposite problem: It's saying the minimum length username for MQTT is 4 characters, but my username is 3.

Seems like a strange limitation to have on the client side of a service, since it's the server side that dictates minimums/maximums for usernames/passwords.